TcMenu/IoAbstraction #10: simultaneously using SwitchInput with switches on several i2c expanders and arduino ports

klaudiusz223, 2018-08-06:
Is it possible to use SwitchInput simultaneously with switches on several i2c expanders and arduino ports?

davetcc, 2018-08-06:
Not as it stands but I could make a multi ioabstraction class that combined several abstractions grouped together. Each one would have a range of pins.

For example 0..100 arduino
100..119 expander 1
120..139 expander 2
Etc 

Let me know if you think that would be useful as I could do that quite quickly.

klaudiusz223, 2018-08-06:
Thank you for your quick response.
To be honest i don’t know. What i’m looking for is some kind abstraction library like yours. I have to implement light and shutter controller based on arduino mega (controllino mega). I need about 70 inputs and 65 outputs. I have only ~ 20 native arduino inputs and ~ 30 native outputs. The rest I/O will be on i2c expanders. I2c expanders will be either input or output (not mixed). I want to have possibility to write some kind of configuration of switches (i.e array of inputs and/or adresses of expanders and their pins ) and configuration of output relays. I want to avoid to write callback function for every switch. I want use your IOAbstraction library and if i could use it for handling switch presses it will be an advantage so i will not need to use modified Bounce library.

davetcc, 2018-08-06:
So there’s a couple of things here.

I think the multi ioabstraction would work and allow up to 256 IO without too many changes. They’d be made up of arduino Ports plus as many io expanders as needed. 

In terms of SwitchInput it defaults to 6 switches, you’d need to change the #define that sets the maximum number of switches in SwitchInput.h to whatever you needed.

I’ll check what that means to RAM usage and comment again later. I’ve never tried it with more than four or five switches and a rotary encoder, but it should be fine with more switches.

davetcc, 2018-08-06:
Right so I'm back at my machine now, where I've got the code checked out..

Each 8574 abstraction, there is a 4 bytes plus the cost of the object, even if 10 were created this should be insignificant on a MEGA2560. the MultiIoAbstraction I've proposed would probably need about maybe another 40 or so bytes to represent those 10. That's probably less than 100 bytes in total for IoAbstraction.

SwitchInput needs around 6 bytes per switch plus any class overhead per switch, plus about another 5 bytes for the controller class,  lets say there's 40 switches this would use less than 350 bytes allowing a extra 2 bytes per object. 

I'll look to try and get something working this week.

davetcc, 2018-08-11:
Now on master, see the new example and also https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/arduino-pins-and-io-expanders-same-time/

To make life a bit easier, switches now also supports PULL UP and PULL DOWN switch logic, new second parameter (default PULL DOWN) to initialise

closed 2018-08-11 by davetcc