TcMenu/IoAbstraction #14: Wire.requestFrom() should not be followed by Wire.endTransmission().

Koepel, 2018-09-01:
In the file "IoAbstraction/src/IoAbstractionWire.cpp", the Wire.requestFrom() is followed by a Wire.endTransmission(). You may remove those Wire.endTransmission(), at the moment at line 67 and 189.
Explanation: [Common-mistakes#2](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes#2)

davetcc, 2018-09-01:
Many thanks, I'll get these incorporated into V1.0.2 and kick off another release later today!

Koepel, 2018-09-01:
That is a little dangerous. The endTransmission() is not a stop.
Most libraries do not break the combination of: beginTransmission-write-endTransmission.

The endTransmission() has a parameter that can send a STOP to the I2C bus or not. Therefor please rename your parameter "bool stopAfterWrite" into, for example, "bool endTransm" or something like that.

davetcc, 2018-09-01:
Agreed, I’ll change it to use a byte enumeration with two values:
LEAVE_TRANSACTION_OPEN and
END_TRANSACTION when I get chance a bit later on.

davetcc, 2018-09-01:
Actually I’ll also make sure that method is private so it can’t be used externally. There’s no valid reason for any external user to call the method anyway.

davetcc, 2018-09-01:
So after all I removed the parameter altogether, I think it's better to have some mild duplication of begin/end than risk it being wrong by obfuscating where it takes place.

I've also slightly optimised the reads by making sure that the ready check is only done once, for the first byte. I'll test this later on 8/32 bit boards and then merge back to master and cut another release. 

Thanks a lot for the help, most appreciated. Let me know if you think there are any concerns with the approach I've taken.

Koepel, 2018-09-01:
As far as I can tell, it is okay. 
Just a few minor details that you probably already know:
- If the I2C address is wrong or the I2C bus is cut, the `waitForReady()` will never return.
- The third parameter for `requestFrom()` is default `true` (guaranteed), so you may omit the third parameter.
- In the function `readIntoMemArray()`, the third parameter of `requestFrom()` is `1`, instead of `true`. That is inconsistant with the others.
- In the function `write32()`, you don't use the mask `0xff` for the first/highest byte. After shifting, the higher bits are zero, but it is converted to a byte anyway. You may use the mask or not for those four bytes, the result is the same for all four bytes. You don't have to worry about that mask. What about a cast to uint8_t ? Some have extra warnings enabled.
- The function `writeAddressWire()` does no longer use the parameter `deviceAddr`.

davetcc, 2018-09-01:
Good point, I’d been using Sloeber for a while but I couldn’t get it working with some boards, so had moved to the Microsoft Arduino plugin for vs code. I need to re-enable the extra warning levels. Up until this move I had pretty much all enabled. I’ll put the changes in and reenable the warnings. I like things to be right. The last one was particularly sloppy on my part! Thanks again! Dave

davetcc, 2018-09-02:
Hi Koepel,

Thanks again for the excellent code review - code is back on master, fully tested on SAMD and AVR! This time I'll leave you to close the issue when you're happy.

Replies to your points:

Point 1: the `waitForReady` function is now bounded, and will only wait about 10millis. If it fails it sets an error variable that can be checked later. This may be the best possible for now. Further, it's hard to test, because Wire itself doesn't handle many error conditions well. I tried removing SDA / SCL and that freezes before hitting my code. However, I confirmed the error by using the wrong address and reading from it, that works. Writes seem to freeze anyway elsewhere (once the buffer's full and it can't write as a best guess).

Point 2: Noted, but I think being explicit helps in this case.

Point 3: Agreed, changed it to `true`

Point 4: Agreed - changed (and warnings turned back to ALL I'd forgotten about that when moving from Sloeber to VsCode plugin)

Point 5: Removed. As above, warnings set up again. 

Dave

Koepel, 2018-09-02:
I'm happy.

closed 2018-09-02 by Koepel