TcMenu/IoAbstraction #20: MultiIo and ioDeviceDigitalWrite

cconde, 2018-09-11:
Hi,

While trying to use Arduino pins and PFC8574, compiler complains when using ioDeviceDigitalWrite and doesn't like receiving the multiIo instance as firs parameter.

I've read documentation and also the reference, and can't guess how to write an output using multiIo.

I'm sure it's a very stupid thing, but I can't see how to get the IoAbstractionRef from the MultiIoAbstraction.

Can you help me?

davetcc, 2018-09-11:
You need to provide a reference to the multi IO by putting an ampersand in front of the variable. It’s only needed for this case of multi IO. 

For example:

ioDeviceDigitalRead(&multiIO, pin);

Let me know if this fixes it.

davetcc, 2018-09-11:
I'll ensure the documentation is cleaned up so that this is made clear.

davetcc, 2018-09-11:
Documentation page on the coders corner already improved. I've adjusted the comments in the MultiIoAbstraction class, but they won't take until I can commit later this week when I add the support for chaining of shift registers.

Many thanks for pointing this out, and hopefully you're up and running with it now.

davetcc, 2018-09-13:
Closing as fixed. Please reopen if the above does not fix the issue

closed 2018-09-13 by davetcc