TcMenu/IoAbstraction #25: ESP8266 Arduino Core: Does not Compile

Phyxius, 2018-11-10:
Compiling the library using Arduino 1.8.7 on the ESP8266 Core 2.4.2 produces the following error:
```
C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp: In member function 'uint8_t I2cAt24Eeprom::findMaximumInPage(uint16_t, uint8_t)':

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: error: no matching function for call to 'min(uint8_t&, int)'

  int currentGo = min(pageSize, offs + len) - offs;

                                          ^

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: note: candidates are:

In file included from c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Phyxius\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src/EepromAbstractionWire.h:10,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:6:

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

  int currentGo = min(pageSize, offs + len) - offs;

                                          ^

In file included from c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Phyxius\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src/EepromAbstractionWire.h:10,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:6:

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

  int currentGo = min(pageSize, offs + len) - offs;

                                          ^

In file included from c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Phyxius\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src/EepromAbstractionWire.h:10,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:6:

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

  int currentGo = min(pageSize, offs + len) - offs;

                                          ^

In file included from c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Phyxius\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src/EepromAbstractionWire.h:10,

                 from C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:6:

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\Phyxius\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Users\Phyxius\Documents\Arduino\libraries\IoAbstraction\src\EepromAbstractionWire.cpp:116:42: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

  int currentGo = min(pageSize, offs + len) - offs;
```

Changing the offending function to explicitly cast arguments to `int` fixes the issue:

```
uint8_t I2cAt24Eeprom::findMaximumInPage(uint16_t destEeprom, uint8_t len) {
	// We can read/write in bulk, but do no exceed the page size or we will read / write
	// the wrong bytes
	int offs = destEeprom % pageSize;
        //-----------------------------Changed line----------------------------------------------
	int currentGo = min((uint)pageSize, (uint)(offs + len)) - offs;

	// dont exceed the buffer length of the  wire library
	return min(currentGo, WIRE_BUFFER_SIZE);
}
```

If anyone else can confirm this issue, I'm happy to submit a PR.

davetcc, 2018-11-10:
Many thanks for reporting, I have an ESP8266 board but had not had a chance to test it on that platform yet. It would be great if you could submit a PR to be added to the async-feature branch:

https://github.com/davetcc/IoAbstraction/tree/async-feature 

Otherwise, I can take the fix and apply it to that branch as I'm working on a significant new feature there at the moment. Let me know which one you prefer.

Thanks again,
Dave.

Phyxius, 2018-11-12:
Submitted a PR for you!

davetcc, 2018-11-12:
Many thanks for the PR. Merged onto the feature branch. Should make it to master soon!

davetcc, 2018-11-24:
This is slightly delayed getting back to master by a bug found during taskManager testing, hopefully will make master tomorrow, followed by a release a few days later.

davetcc, 2018-12-01:
closed - code is now on master, scheduled for 1.2 release

closed 2018-12-01 by davetcc