TcMenu/IoAbstraction #31: PullUp logic per pin/expander

klaudiusz223, 2018-12-10:
At the beginning thank you for implementing multiIO feature #10. I haven't had time to test your library before. Now i did some tests and i must admit that it is very useful. For now i encountered  one problem with pullUpLogic. Switches on native arduino  pins in my case are pull down while these on on expanders are pull up. I had to do ugly trick with SwitchInput::runLoop()
```
		// if the switches are pull up, invert the state.
		if(isPullupLogic() && keys[i].getPin() >= 100 ) {
			pinState = !pinState;
		}
```
Is it possible to do this in more elegant way?
Possible solution could be implemeting per port/expander pullUp logic or using virtual pins which state is returned by function.

davetcc, 2018-12-14:
Possibly the easiest way to do this is to create a NegatingIoAbstraction. That way if some of the devices had different switch requirements, the negating abstraction could delegate something like:

`IoAbstractionRef ref = new NegatingIoAbstraction(ioFrom8574(0x20));`

davetcc, 2018-12-15:
I've had a first go at an abstraction that negates the read and write operations. It should be possible to use this whenever both pull up and down are needed at the same time.

Take a look at NegatingIoAbstraction.h

davetcc, 2018-12-15:
As this one is a bit more contentious, I'll let you close if this actually does what's needed.

klaudiusz223, 2018-12-18:
It works as expected. Thank You!

closed 2018-12-18 by klaudiusz223

davetcc, 2018-12-18:
Great, thanks for confirming and thanks a lot for the changes in the PR too, much appreciated!