TcMenu/IoAbstraction #35: Questions

klaudiusz223, 2018-12-31:
Hi,
please look at example below. I have some questions regarding library behavior.

1. Wouldn't be better not to set pin states during setting pin direction by PCF8574IoAbstraction::pinDirection and leave it to the developer who uses library?
Similar with SwitchInput::addSwitch - not setting pin dirrection?
2.  PCF8574 output pins can't be read.   Only INPUT, INPUT_PULLUP are configured to read.
 is it intentional?  It's not compatibile with native arduino pins
 https://github.com/davetcc/IoAbstraction/blob/e98f2129890f60e639f663c4e68a2c0f419b8866/src/IoAbstractionWire.cpp#L21-L22  
3.  Now it's not possible to mix switches interrupt driven (on PCF8574) and polled (on arduino natives port). Would it be an advantage to have such functionality (for example performance advantage)? 

```
#include <Arduino.h>
#include <IoAbstraction.h>
#include <IoAbstractionWire.h>
#include <NegatingIoAbstraction.h>
#include <Wire.h>

#define INPUT_EXPANDER 0x38
#define OUTPUT_EXPANDER 0x3B

MultiIoAbstractionRef multiIo = multiIoExpander(100);

void onSwitchPressed(uint8_t key, bool held) {
  if (!held) {
   
    uint8_t state = ioDeviceDigitalReadS(multiIo, 108); //  OUTPUT PCF8574 pins can't be read 
    //  only INPUT, INPUT_PULLUP are configured to read.
    //  is it intentional?
    //  it's not compatibile with native arduino pins
    //  https://github.com/davetcc/IoAbstraction/blob/e98f2129890f60e639f663c4e68a2c0f419b8866/src/IoAbstractionWire.cpp#L21-L22
    //  
    
    Serial.println(state);
    state = ioDeviceDigitalReadS(multiIo, 2); // Native arduino pin can be read
    Serial.println(state);
    Serial.println();
    ioDeviceDigitalWriteS(multiIo,  2, !state);
    ioDeviceDigitalWriteS(multiIo,  108, !state);

  }

}

void setup() {
  Wire.begin();
  Serial.begin(9600);

  multiIoAddExpander(multiIo, new NegatingIoAbstraction(new PCF8574IoAbstraction(INPUT_EXPANDER,0xFF)), 8);

  for (uint8_t i = 0; i < 8; i++) {
      ioDevicePinMode(multiIo, 100 + i,INPUT);
  }

  // multiIoAddExpander(multiIo, new NegatingIoAbstraction(new PCF8574IoAbstraction(OUTPUT_EXPANDER,0xFF)), 8);
  multiIoAddExpander(multiIo, new NegatingIoAbstraction(ioFrom8574(OUTPUT_EXPANDER,0xFF)), 8);
  // without negating
  // multiIoAddExpander(multiIo, (new PCF8574IoAbstraction(OUTPUT_EXPANDER,0xFF)), 8);

  for (uint8_t i = 0; i < 8; i++) {
      ioDevicePinMode(multiIo, 100 + 8 + i,OUTPUT);
      // I need to set pins LOW  
      // theoretically they are set LOW by PCF8574IoAbstraction::pinDirection(uint8_t pin, uint8_t mode),
      // but it's invoked before NegatingIoAbstraction
       ioDeviceDigitalWrite(multiIo,100 + 8 + i,LOW);
  }

  switches.initialise(multiIo, false);

  switches.addSwitch(64, onSwitchPressed);
  switches.addSwitch(101, onSwitchPressed);

  
}

void loop() {
  taskManager.runLoop();
}

```

davetcc, 2019-01-06:
Hi there, sorry for the delay in responding. The questions are quite complex and needed me to sit down and think a while before responding. There's a general need for a lot of care changing anything around BasicIoAbstraction now, as this library is being used by a lot of people. Any changes would have to be completely backward compatible.

For point 1: changing the interface / contract is not possible now, there are too many people using the PCF8574 support. Why not extend the class PCF8574IoAbstraction (IoAbstractionWire.h) to meet your needs?

For point 2: this is an optimisation that can be easily fixed by adding a way of setting the read flag to true. I'll add a setReadOverride() method. Then you'll be able to read on ports configured OUTPUT. This I'll put into the next release.

For point 3: this is quite a complex answer, basically, the performance gains come from two sources. Firstly, from not having code polling button states, and secondly, in the case of i2c, not having to read the bus every 20 milliseconds. My recommendation here is not to use interrupts for now. Maybe in a future IoAbstraction release, I could try and implement something that only read from the PCF8574 if there had been an interrupt since the last read. To be honest, it would be low priority at the moment, but I'd gladly accept a PR :)

Dave.

klaudiusz223, 2019-01-19:
Thank You, The answers are sufficient for me. Personally I'm not rely on default pin settings done by library, so at least for me, changes in negating are not important but of course You have to assess yourself.

davetcc, 2019-01-22:
I agree with you that there’s a problem with the negating abstraction. I’ve got an idea.  To make the pinmode call use the base version of write directly. 

I think that will work.

davetcc, 2019-02-19:
Closed this issue and created a new issue to represent the outstanding question on negating IO.

closed 2019-02-19 by davetcc