TcMenu/IoAbstraction #38: Method cancelTask kills also all next tasks

martin-klima, 2019-02-01:
Hi, first many many thanks for this library. I tested the TaskManager for many hours and it looks like there is some bug in cancelTask method.

## Sample code.
Let's have 4 periodic tasks for blinking specific LED. Let's have 5th task which will kill specific periodic task after some time.

```
    taskId1 = taskManager.scheduleFixedRate(1000, [] { toggleLed(0); });
    taskId2 = taskManager.scheduleFixedRate(1000, [] { toggleLed(1); });
    taskId3 = taskManager.scheduleFixedRate(1000, [] { toggleLed(2); });
    taskId4 = taskManager.scheduleFixedRate(1000, [] { toggleLed(3); });

    taskManager.scheduleOnce(4000, [] {
        // Kill a specific task after 4 secs.
        taskManager.cancelTask(taskId2);
    });
```
## Results:
`taskManager.cancelTask(taskId1)` will cancel taskId1 + 2 + 3 + 4
`taskManager.cancelTask(taskId2)` will cancel taskId2 + 3 + 4
`taskManager.cancelTask(taskId3)`  will cancel taskId3 + 4
`taskManager.cancelTask(taskId4)`  will cancel taskId4

## Possible solution
I'm really not C++ programmer, my C++ knowledge is very low. I tried several experiments and this update seems to work in all my tests. 

```
void TimerTask::clear() {
	executionInfo = 0;
	callback = NULL;
	// Removing the next row seems to fix the task canceling problem.
	// next = NULL;
}```

davetcc, 2019-02-01:
Many thanks for reporting this. I’ll look into this as soon as possible.

davetcc, 2019-02-22:
I've got a unit test can re-create this consistently now, will fix for 1.4. Thanks again for providing the above test case, I turned it into an extra unit test that will be run after any change to task manager.

davetcc, 2019-02-22:
Fixed on master with unit test added to test suite.

closed 2019-02-22 by davetcc