TcMenu/IoAbstraction #40: Support request: How to use timer function with parameter?

martin-klima, 2019-02-17:
Hi, is it possible and how to use timerFunction with a variable as parameter?

Example:
```
// This works.
 taskManager.scheduleFixedRate(1000, [] { toggleLed(3); });

// How to call it with variable like this?
byte ledNumber = 3;
 taskManager.scheduleFixedRate(1000, [] { toggleLed(ledNumber); });
```
I have tried probably all possibilities but with no luck.
Thank you for any help.

davetcc, 2019-02-18:
Hi there, this is more of a C++ question but yes it is possible. When you write the function shorthand [] { ... } what you are declaring an inline function essentially. 

The square brackets are for capturing variables, but you should question if it’s a good idea to capture a local variable like that. There is a lot of compiler magic going on behind that and it may affect memory usage on the device. Check carefully before baking a lot of them into your code. 

Take a look at capture clause in this doc:

https://docs.microsoft.com/en-us/cpp/cpp/lambda-expressions-in-cpp?view=vs-2017

davetcc, 2019-02-18:
I’ll leave you to close this once you’re happy it’s answered.

martin-klima, 2019-02-18:
Thank you for your kind answer. I read the documentation page yesterday. I'm noob in C++ and I've tried everything without success. I think there can be many reasons why to call your Taskmanager scheduler functions with parameter and it would be nice to have some example in docs or Readme. Let me show the minimal example code. 
```
/**
* Minimal sample of problem.
*/

#include <IoAbstraction.h>

void setup()
{
    // Start WaterDipatcher for box 1 in 5 secs.
    scheduleWaterCheck(1, 5);
}

/**
 * The plan is to call this function with params and schedule waterDispatcher by boxId.
 */
void scheduleWaterCheck(byte boxId, unsigned int scheduleTime) {
    // @todo Solve problem with variable in parameter.

    // Use a wrapper and avoid any parameter - works.
    taskManager.scheduleOnce(scheduleTime, waterDispatcherWrapper_1, TIME_SECONDS);

    // Use a fixed parameter - works:
    taskManager.scheduleOnce(scheduleTime, [] { waterDispatcher(1); }, TIME_SECONDS);

    // Use variable as parameter - doesn't work:
    // error: 'boxId' is not captured
    taskManager.scheduleOnce(scheduleTime, [] { waterDispatcher(boxId); }, TIME_SECONDS);
    
    // This doesn't work:
    // warning: invalid user-defined conversion from 'scheduleWaterCheck(byte, unsigned int)::<lambda(byte)>' to 'TimerFn {aka void (*)()}' [-fpermissive]
    taskManager.scheduleOnce(scheduleTime, [&](byte boxId) {
        waterDispatcher(boxId);
    }, TIME_SECONDS);
}

/**
 * Wrap function with parameter to function w/o parameter.
 */
void waterDispatcherWrapper_1()
{
    return waterDispatcher(1);
}

void waterDispatcher(byte boxId) {
    // Do some actions with boxId here...;
}

void loop()
{
    // This is all we should do in loop when using task manager.
    taskManager.runLoop();
}
```
Can you please provide some working example of calling scheduler function with parameter please? Or is it a totally wrong way? Am I thinking wrong about it?
Many thanks!

davetcc, 2019-02-18:
Agreed, I’ll add an example in the next day or so that shows a few ways of passing parameters into task manager lambda functions.

davetcc, 2019-02-19:
Basically, I've thought about this today in more detail. If you try to capture, you're essentially capturing from the local stack to a function that will be called when that stack doesn't exist any longer. I think the best way of passing such parameters is probably to create a class (or some kind of global variable to store the state).

I've brought forward a plan to support calling back classes, where you can define a simple class with all the state you need and have the scheduler call it. This I think should fulfill most of what you want it to do. I'll get this committed shortly along with examples, see #42

martin-klima, 2019-02-19:
#42 looks very nice and it could solve all my needs. I will pull latest update and test it. Many thanks!

davetcc, 2019-02-20:
Great, just be aware that’s still not been fully tested yet. I tested it on my MEGA board yesterday and ran it for a few hours.

I’ll write some unit tests for it and also to try and find the remove bug and add them to the suite. They should both be released in 1.4 shortly.

davetcc, 2019-02-21:
Could you please close this issue now? You can put any feedback onto #42 if you think additional functionality is needed.

davetcc, 2019-02-22:
I'll close this one now, any issues put it onto #42. I noticed a compile error for SAMD with the new task manager code and fixed it up just now.

closed 2019-02-22 by davetcc