TcMenu/IoAbstraction #43: Use multiple rotary encoders

K4stor, 2019-02-24:
Hi

I am using a PCF8574 and connected more than one rotary encoder to it.
Than I init them like this
```
  Wire.begin();
  switches.initialiseInterrupt(ioFrom8754(0x20, 2), true);
  presetEncoder = new HardwareRotaryEncoder(0, 1, onPresetEncoderChange);
  param1Encoder = new HardwareRotaryEncoder(2, 3, onParam1EncoderChange);
  param2Encoder = new HardwareRotaryEncoder(4, 5, onParam2EncoderChange);
  param3Encoder = new HardwareRotaryEncoder(6, 7, onParam3EncoderChange);

  switches.setEncoder(presetEncoder);
  switches.setEncoder(param1Encoder);
  switches.setEncoder(param2Encoder);
  switches.setEncoder(param3Encoder);
```

The problem is that I can only set ONE encoder as active. So only the last set works. 
Is there another way to do this ?

davetcc, 2019-02-24:
Unfortunately at the moment there is only support for one encoder.

Adding more than one needs a Different way of handling interrupts. If we take switches out of the equation and just manage the extra ones manually it may be possible with a few small library changes that don’t break backwards compatibility.

The bigger issue is that all rotary encoders need interrupt pins for pin A using this library. On a 32 bit SAMD board that’s not a big problem but on AVR that could become a problem.

K4stor, 2019-02-24:
Thanks... well I am having a look as well. And if I understand it corretly the interrupt is marked, and in the next run loop the encoder is evaluated, and if there is an increment, you fire an event.
Isn't it just needed to have a list of encoders and process them one by one ?

Regards

davetcc, 2019-02-24:
That is pretty much correct. 

Actually, the biggest issue is keeping backward compatibility and memory usage low enough for Uno use. Now a days there are a lot of users of the library. 

I think some kind of linked list could keep references to the next encoder without costing much in memory. 

There could however be unforeseen issues with having many encoders on one interrupt, I’ve not tried it before. I doubt it though.

K4stor, 2019-02-25:
Ok... I ran some tests... and the key is to use
RotaryEncoder* encoder[8] and add a overload to the setEncoder method with an index as argument.
The old method would still work and just use encoder[0] as default. 

then I modified the loop
```

void onSwitchesInterrupt(__attribute__((unused)) uint8_t pin) {
	if(switches.isInterruptDriven() && !switches.isInterruptDebouncing()) {
		checkRunLoopAndRepeat();
	}

  for(int i = 0; i < 8; i++) {
		if(switches.encoder[i]) {
			switches.encoder[i]->encoderChanged(); // TODO
		}
	}
}
```

so it checks interrupts.
So legacy code would still work. No linked list is needed.
If you like I could create a pull request and you have a look. If you don't like it you could still dismiss it. Cool ?

davetcc, 2019-02-25:
Many thanks for the PR. I've merged it back to master now and made a few small changes to integrate it. I ran most of the encoder examples, at least for the single case.

I lowered the default number of extra encoders to 4 and made it configurable with a define MAX_ROTARY_ENCODERS. Just edit the SwitchInput file if you need more than 4.

Please, could you pull down master and see if it works for you, closing this issue if all is good. Otherwise commenting on anything not working with it.

Thanks again, 
Dave.

K4stor, 2019-02-25:
Damn somehow I did a copy and past mistake and the array size is wrong... creating a fix RIGHT NOW... SORRYY !

davetcc, 2019-02-25:
Don’t worry I fixed that as I brought it over. No problem there.

K4stor, 2019-02-25:
Never mind I see you did use MAX_ROTARY_ENCODERS so its all good...phew 👍

davetcc, 2019-02-25:
I lowered it down to 4 encoders by default with an easy define to increase for larger numbers. Should be good for your case now.

davetcc, 2019-02-25:
Thanks again for the help with this.

closed 2019-02-25 by K4stor