TcMenu/IoAbstraction #59: Any Chance this can support CD4021 Shift Registers?

tekadept, 2019-07-08:
I see in the Shift Register section it only mentions 74HC165 for input. Can it support CD4021 Input Registers as well?

tekadept, 2019-07-08:
Just trying to do it, and it with a single shift register, and it ony seems to fire on pin p1-8 and records all pins as firing. if that makes sense.

davetcc, 2019-07-08:
Thanks for the issue, I’ll take a look at the data sheet and see what’s possible.  I’m not actually sure what the difference between these devices is.

tekadept, 2019-07-09:
Thanks, I have a CD4021 working with this code,
https://github.com/InfectedBytes/ArduinoShiftIn
For that code I just had to remove any reference to the ploadpin for it to work, if that helps.

davetcc, 2019-08-17:
Test the 74HC165 as part of the 1.5.0 release to ensure it's working. There are explanations on the web how to use this device in place of the 74HC165 device. I suspect there is an issue in the input shift register support rather than with this device. To be determined..

davetcc, 2021-02-08:
See #110 which has code to fix the issue, we'll look to integrate it into the next release.

closed 2021-02-08 by davetcc