TcMenu/IoAbstraction #60: Possibility of Interrupts on Shift registers in the Future?

tekadept, 2019-07-08:
Just wondering if its likely to ever be supported for interrupts on Shift registers in the future? this would be extremely powerful indeed.

tekadept, 2019-07-08:
Are Event handles ie, add switchsupported by Shift Registers? as I can't seem to get it to work..
simplistic code..
IoAbstractionRef shiftRegister = inputOnlyFromShiftRegister(clockPin, dataPin,latchPin,1);

void onClicked(uint8_t pin, bool heldDown) {
  //Serial.print(heldDown);
  Serial.println(pin);
}

void setup()
{for (int i = 0; i < 8; ++i) {
		ioDevicePinMode(shiftRegister, i, INPUT);
}
switches.initialise(shiftRegister,false);
switches.addSwitch(6,onClicked);
switches.addSwitch(7,onClicked);

void loop()
{
  ioDeviceSync(shiftRegister);
  taskManager.runLoop();
  }

This only fires when I push Button on p1-8 on my CD4021 shift register and prints 6 7

davetcc, 2019-07-09:
Hi there thanks for the issue. Ive got a shift register to hand so I’ll build the circuit and test when I get a bit of time, probably later this week.

To be honest Interrupts are not possible on shift registers. The chip doesn’t have an interrupt line.  If you need interrupts I’d consider one of the i2c expanders.

davetcc, 2019-07-09:
Also switches will do the IoSync for you, it can be removed from the loop method. You can also remove the pinmode configuration from setup. That’s also done when you add a switch.

Don’t forget that unlike Arduino pins and IoExpanders such as off 8574 and MVP 23017 cannot do INPUT PULL-UP so you’ll need to add your own pull up or down resistors. 

I’ll feedback further after I’ve assembled the circuit later this week.

davetcc, 2019-08-10:
This question is going to be closed out:
Interrupts cannot be supported on shift registers as there is not interrupt support on the chip.
The other part is already covered in the other related issue #59

closed 2019-08-10 by davetcc