TcMenu/IoAbstraction #62: Support Analog Events

GuidoLeibbrand, 2019-08-15:
I`m new in this Framework and like the idea to use tasks instead of loops.
So it was possible to write interrupt free code. This is necessary for my project.

In the first step i used digital pins and it worked great.
Now i have to expand my project and listen to a analog pin value. If the analog value between 0-200 then raise an event an execute a methode. 
I`m not found support for this topic.

I see there is support for RotaryEncoder but there are two pins requestet in 
```setupRotaryEncoderWithInterrupt(encoderAPin, encoderBPin, onEncoderChange);```. But i think it is not fit for my problem.

Can you tell me is this feature on the roadmap?

br,
Guido

davetcc, 2019-08-16:
At the moment there is an analog abstraction within the package but it has no interrupt support. Also, although there’s a plan to support i2c and SPI analog devices it currently only works with Arduino pins.  

It’s not particularly well documented at the moment because it’s very new. One of my next tasks is to tidy up the documentation for IoAbstraction because it’s got untidy. 

I’ll see how much work it is to add interrupt support in the next few days and update this issue. It may be quite simple in which case I’ll add it to the next release or the one after.

davetcc, 2019-08-17:
I'll add this to the 1.5.0 release pipeline as it will take quite a bit of work and thinking through. It will have to work in some way across most common Arduino boards and mbed boards. The plan is already there to do it, with a fallback to have a software implementation.

For now you could easily emulate this behaviour in software with task manager. Just register a task that checks the analog pin fairly frequently, and raises a new task if the condition is met. See the analog abstraction for the starting point of where the extra functionality.

davetcc, 2020-07-06:
Bumped to 1.6, not enough bandwidth to do this in 1.5.

davetcc, 2020-08-06:
The next version of TaskManager will have general-purpose events built into it. On top of this it will be trivial to implement events for any task, interrupt or polled, let alone just analog events.

davetcc, 2020-08-25:
It was a long road to get here, and maybe you've moved on and used a different library by now. But a generic event framework will be in the next version, with a specialization for analog events. Due for release soon. See the commit or the ioa-taskmgr-refactor if interetsted.

closed 2020-09-01 by davetcc