TcMenu/IoAbstraction #65: multiIo uncertainty

psgcooldog, 2019-09-19:
Dave,

Love this library!  Had things working beautifully with a single MCP23017, but when I added a second, and tried to use the MultiIoAbtraction things did not go so well.  It's not working at all, and crashes the ESP32 quite frequently.  Here's my initialization code:

#include <IoAbstraction.h>
#include <IoAbstractionWire.h>

MultiIoAbstraction multiIo(0);    // no Arduino pins
IoAbstractionRef io23017_0 = ioFrom23017(MCP23017_0_ADDR, ACTIVE_LOW, INTA0);
IoAbstractionRef io23017_1 = ioFrom23017(MCP23017_1_ADDR, ACTIVE_LOW, INTA1);

void initializeIo() {
  multiIo.addIoExpander(io23017_0, 16); 
  multiIo.addIoExpander(io23017_1, 16); 
  switches.initialiseInterrupt(io23017_0, true);
  switches.initialiseInterrupt(io23017_1, true);
}

and then set up the switches:

void initializeFs() {
  for (int i = 0; i < nFs; i++) {
    switches.addSwitch(Fs[i].inBit, processFs);
    switches.onRelease(Fs[i].inBit, processFs);
}

Any hints?  I've looked at your examples, but none of them shed any light on my problem.

davetcc, 2019-09-19:
Many thanks for the compliment. 

By default switches is only set up for 5 switches. Do you need to increase the number or switches?

If so go to the library directory and view src/SwitchInput.h

You can either define MAX_KEYS yourself before the build starts or you can edit the file and set it to your requirements. Memory increase for more switches will be irrelevant on an ESP or other 32 bit board. 

Snippet from SwitchInput.h:

#ifndef MAX_KEYS
#define MAX_KEYS 5
#endif // MAX_KEYS defined

Let me know if that fixes it. Otherwise I have an ESP32 and several expander ICs. 

Thanks
Dave

davetcc, 2019-09-19:
See #66 to prevent more switches than are configured being added.

psgcooldog, 2019-09-19:
I did have it set to 32, which is what I'm using, but just in case there was a boundary condition error, I'm testing at 128 now.

No change.

psgcooldog, 2019-09-19:
This is yet another PCB prototype, to which I've now added a second MCP23017, as well as a microSD slot.  The I2C scanner does show both expanders as answering, and the first MCP23017 was working fine (on this specific board) before I tried to set up the multiIO thing to add the second one into the software.

psgcooldog, 2019-09-19:
Does my init code look good?  Any problems with setting up zero Arduino pins in the initial constructor?  How about the way I'm setting the switches to use interrupts?  I just had to guess at what was the right way to do that...

psgcooldog, 2019-09-20:
A hint: my SDcard interface is on the following SPI pins:
// SPI stuff for microSD
#define SPI_SCLK       18
#define SPI_MISO       19
#define SPI_MOSI       23
#define SPI_SS          5

When I insert or remove the card, I'm getting indications of switch closures that *should* only be coming from the footswitches attached to IO expanders.  I'm trying to only use the two IO expanders and their associated INT pins through the library, the rest are I2C, serial MIDI ports, or analog inputs.

#define INTA0         27
#define INTA1         17

davetcc, 2019-09-20:
I’ve never tried using it with 0 pins as there is an Arduino pin abstraction on those pins. Zero may work but was certainly not tested, maybe try setting it to 10 or something small. It still leaves 244 pins to use!

I did notice this:

switches.initialiseInterrupt(io23017_0, true);
switches.initialiseInterrupt(io23017_1, true);

Call initialise interrupt only once on switches and pass it the multiIo reference instead. Then ensure that you use the PIN numbers from the multiIo.

switches.initialiseInterrupt(&multiIo, true);

In terms of the instability on inserting the SD card I’m not sure. Have you any floating pins? Are the Arduino interrupt pins definitely pulled up? Maybe check with a scope. 

Also, The MCP23017 needs at least a 0.1uF very near the power rails to avoid instability.

davetcc, 2019-09-20:
Something else: how long is the i2c bus wiring and what pull-up resistors are on the end of it?

psgcooldog, 2019-09-20:
If I start with multIo(10), but never do anything to those pins 0-9 with the library, does it touch those pins at all?

I'll fix the initialiseInterrupts and test in the morning, if that doesn't work I'll go with adding some Arduino pins and try that.

The hardware is good, the bus is only a few inches, and pulled up with 10K.  Remember, this hardware works just fine if I just use the single IO expander w/o multiIO.  All the chips have .1uF bypass caps within 1/4" or less, and there's plenty of power and ground plane copper.

I don't think the action I was seeing while inserting the SDcard was instability; I *think* it was your library reading pins on the "Arduino" and reporting them as the pin numbers I expected to see from the first IO expander.  The switches that are connected to that expander do nothing when I use the multiIO setup, but work just fine with only the single expander configured directly.

It was a bit crazy today; I also discovered that sometime in the last week or so my custom SysExMaxSize setting for the MIDI library just somehow stopped working.  No clue why.  In the end I just shrugged, dialled up the default in the library and recompiled.  I'm also fighting a mysterious crash-bug incompatibility between the ESPAsyncWebserver and Arduino-BLE-Midi libraries.

davetcc, 2019-09-20:
Thanks for confirming the i2c setup etc. I think we can rule out that class of issues. 

MultiIo:
setting multiIo to start at 10 will just mean that you can use pins 0..9 if you wish and they would map to arduino pins. Your expanders would then be on pins 10..25 and 26 onward in the order they were added.

Before the multiIo was not being used by the library. So the last call to initialise would have set your second expander only.

Let’s see where we are after you make the changes. If there’s still issues I’ll try and test a similar scenario this weekend. I have an ESP32 to hand. 

Unrelated question, I’ve been using visual micro for most things but am interested in platform IO as I need to support mbed and other platforms. How do you find it and what ide do you use?

psgcooldog, 2019-09-20:
SUCCESS!  the key was fixing the switch.initialiseInterrupts call.  At first blush, it seems to be working handily with zero Arduino pins defined.  I *might* change that in the future, but most likely not.  Also, both expanders are working.

I am absolutely *THRILLED* with both platformIO and Visual Studio Code.  It makes for a *real* development environment, instead of the toy Arduino one.  I did use Arduino for a couple of years, but got frustrated by it's lack of ... well, pretty much everything, as well as it's horrifically long compile times.

Make the switch.  It'll take you an hour to get up and running and productively working on an imported Arduino project.  Because it uses make, and you can easily split your code up properly, build times are *drastically* shorter than in Arduino.  I'm hooked, and would never go back.

I'm going to open another issue for an enhancement request that would clean my code up a whole lot.  Thanks so much for your help.

closed 2019-09-20 by psgcooldog