TcMenu/IoAbstraction #67: Add optional arguments to TaskManager schedule requests

psgcooldog, 2019-09-20:
Being able to pass a variety of arguments to scheduler requests would be quite valuable:

uint8_t TaskManager::scheduleOnce(uint16_t when, TimerFn timerFunction, TimerUnit timeUnit = TIME_MILLIS, void *arg1 = NULL, void *arg2 = NULL);

uint8_t TaskManager::scheduleOnce(uint16_t when, TimerFn timerFunction, TimerUnit timeUnit = TIME_MILLIS, uint16_t arg1 = 0, uint16_t arg2 = 0);

uint8_t TaskManager::scheduleOnce(uint16_t when, TimerFn timerFunction, TimerUnit timeUnit = TIME_MILLIS, void *arg1 = NULL, uint16_t arg2 = 0);

etc, etc, etc.

Use case: I have an array of footswitches with associated LEDs.  When a switch is depressed, I want to flash the LED for 350ms.  I turn it on, and schedule a task to turn it off in 350ms, but there's no way to tell that task which LED I want off.  My workaround was to define an array of functions, one for each LED.  Messy...

davetcc, 2019-09-22:
There is a capability for this already. It's done by extending the Executable class and passing a reference to that class instead of a function to the schedule methods.  You just keep whatever data is needed in your class, and it's available to you when your exec() method is called.

Take a look at [https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/] in the official documentation. There is a section on using the Executable class. The task manager examples include usage of this too.

davetcc, 2019-09-22:
I've just updated the documentation to include a full example on the above linked page. Also, I'm planning to add this support to switches as well, so that switches can call an Executable as well. At the moment it cannot.

psgcooldog, 2019-09-22:
Thanks Dave!  I must confess that I'm an old C programmer, so I don't naturally think in Classes, Objects, overrides, etc.

Got it working.  Implemented it as IntegerExecOnce(void (*func)(int), int val) , and it commits suicide after the function runs.

closed 2019-09-22 by psgcooldog