TcMenu/IoAbstraction #70: Enhancement - allow devices on multiple I2C buses

psgcooldog, 2019-09-30:
Dave,

It just occurred to me that this might be a showstopper with the prototype boards that arrive this week.  I have moved the MCP23017 for the footswitch off the main board, and on to the new footswitch board along with three LED drivers (TLC59116s).  These are on the *second* I2C bus , not the standard one.  I still have one MCP23017 onboard, on the first I2C bus.

I don't see any way of handling that in lib.

Paul

davetcc, 2019-09-30:
Actually this is not too difficult to fix. Let me take a look when I get a few minutes and get back. I think I could make it possible to set the wire class to use on the 23017 and 8574 abstractions.

psgcooldog, 2019-09-30:
That would be awesome.  My apologies for the false alarm on the other issue; I had edited that #define several times, but then refreshed the library to be certain I was getting your latest changes, and forgot.  I'd just gotten into bed when I sat bolt upright and said "son-of-a-bitch!  I think maybe ..." and ran back downstairs to check.

davetcc, 2019-10-01:
I've had a go at this, basically whenever you configure a IoAbstractionRef you can provide a TwoWire* pointer as the last parameter:

`IoAbstractionRef ioFrom23017(uint8_t addr, Mcp23xInterruptMode intMode, uint8_t interruptPin, TwoWire* wireImpl = &Wire);`

And for a PCF8574:

`IoAbstractionRef ioFrom8574(uint8_t addr, uint8_t interruptPin = 0xff, TwoWire* wireImpl = &Wire);`

And for i2c EEPROMs the constructor now accepts a wire pointer in the last parameter:

`I2cAt24Eeprom(uint8_t address, uint8_t pageSize, TwoWire* wireImpl = &Wire);`

psgcooldog, 2019-10-01:
Looks great.  I may not be able to test it for a while; looks like the latest rev of GigGadget is stuck in China until their National Day holiday is over.  I've had a DHL tracking number for three days, but it hasn't been picked yet. :(

davetcc, 2019-10-12:
Hi Paul,

I hope you don't mind but I'm going to move this into complete today, it works for all the examples and test cases that I have. If it turns out that it doesn't work with this case on ESP when you get the boards, you can re-open it and I'll fix it up.

Dave

closed 2019-10-12 by davetcc