TcMenu/IoAbstraction #71: Enhancement - use clues about the running environment to choose better resource limits

psgcooldog, 2019-09-30:
It ought to be possible to choose better values for MAX_KEYS and TASK_SIZE by looking at other defined constants.  Perhaps sizeof(char*) ?

davetcc, 2019-09-30:
It’s really only the uno that’s driving the super small values now. For tasks I could set to 20 on 32 bit boards, 10 on mega and keep the same for uno.

But for switches it may be easier to make it reallocate the array on mega and 32 bit boards in segments of maybe 16. Maybe allowing up to 3 reallocations. 

This needs some thought though.

psgcooldog, 2019-09-30:
In that case just check for the Uno.  I like the sound of allowing multiple reallocs.  Not sure why you'd want to limit that; either the user really needs what they're asking for, or they have a bug.  Either way, they'd still have something to fix.

davetcc, 2019-10-01:
I was thinking uno start at 6 like now, otherwise start at 16, either way if those get used, reallocate by doubling each time and copying over then deleting

psgcooldog, 2019-10-01:
Sounds good to me.

davetcc, 2019-10-01:
This needs a bit more reworking, so I'll make this change in the next major release. Hopefully, that just leaves just this one release (1.4.10) to do the manual adjustments!

davetcc, 2020-05-18:
Switches now uses the new lightweight collection and its list of managed buttons grows as needed. Taskmanager has now been modified so that findFreeSlot will reallocate if the collection doesn't fit. It does mean that it temporarily needs more RAM, but this should be fine for most cases.

davetcc, 2020-06-24:
Task manager and switches now increase in size as needed. There are no longer hard limits. The only hard limit now in the library is the number of rotary encoders, which defaults to 4.

davetcc, 2020-06-24:
to release in 1.5

closed 2020-06-24 by davetcc