TcMenu/IoAbstraction #72: 4x7 keyboard hangs with certain keys

hjf, 2019-10-24:
I'm using a 4 row, 7 column keyboard. For the rows I'm using Arduino pins, and for the columns I'm using a shift register. I've declared my keyboard as:

```
IoAbstractionRef multiIo = multiIoExpander(100);
const char pgmLayout[] PROGMEM = "abcdTrq147ELtw2580Cye369NKiu";
KeyboardLayout keyLayout(4, 7, pgmLayout);
MatrixKeyboardManager keyboard;
 multiIoAddExpander(multiIo, outputOnlyFromShiftRegister(SR_CLK, SR_DAT, SR_LAT), 64);

  keyLayout.setColPin(0, 133);
  keyLayout.setColPin(1, 134);
  keyLayout.setColPin(2, 135);
  keyLayout.setColPin(3, 136);
  keyLayout.setColPin(4, 137);
  keyLayout.setColPin(5, 138);
  keyLayout.setColPin(6, 139);

  keyLayout.setRowPin(0, 7);
  keyLayout.setRowPin(1, 8);
  keyLayout.setRowPin(2, 9);
  keyLayout.setRowPin(3, 10);

  keyboard.initialise(multiIo, &keyLayout, &myListener);
  keyboard.setRepeatKeyMillis(850, 350);
```
This mostly works, but the program "hangs" if I use the keys from the forth row (3, 6, 9, E, K..).
This didn't happen when I was just testing the program, but it started happening when my program became larger. I think the problem may be in the alignment of pgmLayout. Maybe it's crossing a byte boundary?

If I remove the rest of my code and leave only the keyboard part, all keys work.

davetcc, 2019-10-25:
Could you confirm the sketch size and board type as I think it is using near versions of the PGM functions. It is possible that a sketch above 64k could cause this.

What I could do is to use the far versions of all the functions when the board is not an uno. I’ll feed back when I’ve had chance to look at this. Probably tomorrow now.

davetcc, 2019-10-26:
Hi there,

I think the problem is here

```    char keyFor(uint8_t row, uint8_t col) {
        if(row < rows && col < cols) return pgm_read_byte_near(&pgmLayout[(row * cols)+col]);
        else return 0;
    }
```
Line 87-90 in KeyboardManager.h within the library. You could try changing `pgm_read_byte_near` to `pgm_read_byte_far` and see if that fixes it.

davetcc, 2019-10-26:
Thanks for reporting this, I've assigned to patch release 1.4.11. Possible workaround above.

hjf, 2019-10-26:
Interestingly enough, the problem went away when I changed the pins from 7,8,9,10 to 7,8,9,A5. I don't understand why. Nowhere else in the program I'm using pin 10. I am, though, using SPI for the nrf24 library but I don't know if that would mess with pin 10. It does use 11,12,13 though. 

My program is about 17k BTW and the  MCU is a mega 328p so it can't be over 64K

davetcc, 2019-10-26:
Interesting but as far as I can tell pin 10 (PB2) has no other function. Does any other library, hardware or or shield you are using indirectly use that pin? Is it completely stable on A5?

hjf, 2019-10-26:
Yes, used it quite a bit with no problems after moving the pin. The only other external library I'm using is https://github.com/nRF24/RF24. but it shouldn't be using pin 10 at all.
Pin 10 is now disconnected and all seems to work, I even added a rotary encoder to the mix.

davetcc, 2019-10-26:
Great let me know if you see any issues. I’ll keep this open for a while. The near / far needs to be looked in to anyway.

davetcc, 2019-11-01:
I’m assuming you’ve not seen this in the past week. I’ll close out the issue but if you see it again please reopen.

closed 2019-11-01 by davetcc