TcMenu/IoAbstraction #73: Rotary encoder acceleration

Express1, 2019-11-05:
I use the buttoRotaryEncoder example for the ioAbstraction library (latest library version, arduino nano).
Is working as expected but I have a question regarding the encoder - when turned really slow will go in a normal sequence like 1,2,3,4. If turned faster will skip - 12,22,32.
So it looks like some sort of acceleration is implemented. Is there a way to turn this off?
I did look a little bit at the docs but I was not able to find anything related.

Thank you,
Radu

davetcc, 2019-11-06:
Yes there is acceleration implemented in the encoder.  Unfortunately it is presently not configurable. I will add this in the next release.  

However it is very easy to change the code temporarily until the next release. 

Just go to src /SwitchInput.cpp and change the whole body of the following method to just return 1. 

HardwareRotaryEncoder::amountFromChange(unsigned long change)

Express1, 2019-11-06:
Based on your reply I modified src /SwitchInput.cpp and is working as expected.
on top#define NOACCEL
....
 
int HardwareRotaryEncoder::amountFromChange(unsigned long change) {#if defined(NOACCEL)<------>// no accel<------>return 1;#else defined(NOACCEL)<------>if(change > 250000 || maximumValue < ONE_TURN_OF_ENCODER) return 1;
<------>if(change > 120000) return 2;<------>else if (change > 70000) return 4;<------>else if (change > 30000) return 6;<------>else return 10;#endif defined(NOACCEL)}

davetcc, 2019-11-07:
Good news I’ll get another release out soon where acceleration is optional.

davetcc, 2020-02-16:
Will be in release 1.4.11 today

davetcc, 2020-02-16:
HW rotary encoder has new function: 

`void setAccelerationMode(HWAccelerationMode mode) { accelerationMode =  mode; }`

This has three values:

`
    /** No acceleration, no matter how fast the encoder is turned */
    HWACCEL_NONE,
    /** The default, accelerates based on how fast the encoder is turned */
    HWACCEL_REGULAR,
    /** Slower acceleration than above is applied */ 
    HWACCEL_SLOWER
`

closed 2020-02-16 by davetcc