TcMenu/IoAbstraction #75: esp8266 exception 28 crash

tablatronix, 2020-02-05:
Anyone have any ideas about this one?

my entry point is new HardwareRotaryEncoder(

```php
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

Exception (28):
epc1=0x40204b20 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01a0
3fffff80:  40203575 000003e8 00000018 3ffee6cc  
3fffff90:  3fffdad0 3ffee664 3ffef60c 402010e7  
3fffffa0:  feefeffe 00000000 3ffee68c 4020298c  
3fffffb0:  feefeffe feefeffe 3ffe84ec 4010129d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
```

## Decoded
```php
0x40106d00: ioDevicePinMode(BasicIoAbstraction*, unsigned char, unsigned char) at ??:?
0x402010f7: setup at ??:?
0x40202498: loop_wrapper() at core_esp8266_main.cpp:?
0x401010b9: cont_wrapper at ??:?
0x4010f000: ?? ??:0
```

Seems to be caused by `switches.getIoAbstraction()`

`IoAbstractionRef getIoAbstraction() { return ioDevice; }`

Not sure what memory access problem is here, maybe something needs an `ICACHE_RAM_ATTR` ?

Anyone got an idea?

davetcc, 2020-02-05:
Are you using the master version? That has some changes for ESP boards.

Could you paste a sketch here that recreates it and I’ll try it with master code base myself.

tablatronix, 2020-02-05:
Yes I am

tablatronix, 2020-02-05:
Ill probably have to add delay(0) to all these taskman while loops to make sure its not a wtd, it doesn't look like one, usually these are illegal memory accesses or overflows

davetcc, 2020-02-05:
I doubt it’s watchdog, unless you have some very long running tasks. I think that it looks to me like something not mapped with the ram cache attribute being used in the interrupt handler.

I’ll look at this code later and respond

davetcc, 2020-02-05:
I’ve looked at the code and think I have the answer. It’s the way that the handler is marking the interrupt flag on the task manager object.

Strange how I’ve got away with this on my esp test boards for so long. But the program is quite small. 

It will need a code fix to mark the interrupt flag differently.

tablatronix, 2020-02-05:
Thanks, I can take a look again and clean up some code, I was messing around with trying to modify

HardwareRotaryEncoder.ino
to use PCF but without interrupts..
I need to use polling instead , 1 because my module doesnt break out int, also because I need to disable interrupts during most of my program but would like to still allow button presses of low priority etc.

davetcc, 2020-02-05:
Had you adjusted the core library code in any way before getting the error above? Just trying to understand the problem better. 

Unfortunately, at the moment it’s not possible to use hardwareRotaryEncoder without interrupts. It could be modified to support this mode of operation, but it would need to poll exceptionally frequently. Maybe even several times a millisecond.

tablatronix, 2020-02-05:
heh no I reverted everything before testing and submitting this, 

Here is what I was doing, I removed the pcf stuff for now.

```C++
/*
 This sketch gives an example of using a rotary encoder along with it's inbuilt push button with
 the switch input facilities. In addition to the encoder, we also connect another switch that
 will be used to 
 
 Switch input is designed to work with the task manager class which
 makes scheduling tasks trivial.

 Circuit / detail: https://www.thecoderscorner.com/products/arduino-downloads/io-abstraction/arduino-switches-handled-as-events/

*/

#include <IoAbstraction.h>
#include <Wire.h>

// The pin onto which we connected the rotary encoders switch
const int spinwheelClickPin = 0;

// The pin onto which we connected the repeat button switch
const int repeatButtonPin = 25;

// The two pins where we connected the A and B pins of the encoder. I recomend you dont change these
// as the pin must support interrupts.
const int encoderAPin = 2;
const int encoderBPin = 3;

// the maximum (0 based) value that we want the encoder to represent.
const int maximumEncoderValue = 128;

//
// When the spinwheel is clicked, this function will be run as we registered it as a callback
//
void onSpinwheelClicked(uint8_t pin, bool heldDown) {
  Serial.print("Button pressed ");
  Serial.println(heldDown ? "Held" : "Pressed");
}

//
// When the repeat button is pressed, this function will be repeatedly called. It's also a callback
//
void onRepeatButtonClicked(uint8_t pin, bool heldDown) {
  Serial.println("Repeat button pressed");
}

//
// Each time the encoder value changes, this function runs, as we registered it as a callback
//
void onEncoderChange(int newValue) {
  Serial.print("Encoder change ");
  Serial.println(newValue);
}

void setup() {

  Serial.begin(115200);

  HardwareRotaryEncoder* myEncoder = new HardwareRotaryEncoder(encoderAPin, encoderBPin, onEncoderChange);
  switches.setEncoder(0, myEncoder);
  switches.changeEncoderPrecision(maximumEncoderValue, 100);

}

void loop() {
  taskManager.runLoop();  
}
```

tablatronix, 2020-02-05:
I was going to figure out how to add manual polling in after I got it working, My current code is polling my encoders at 50ms with ok reads, I was looking to switch it over to an expander as I need more pins. Also I have a ton of these lcd backpacks ( with no int breakout ) and was hoping to fix them in software rather than bodging an pin on them

tablatronix, 2020-02-06:
Ok I found my problem,
I either missed something in the docs, or commented something out.

I was not doing anything to initialize SwitchInput, so ioDevice was blowing up..

Ok back to trying this with something other than
`ioUsingArduino`

tablatronix, 2020-02-06:
I was following the readme `RotaryEncoder - hardware and button emulation, even available with i2c IO expanders` and it did not mention `switches.initialise`

closed 2020-02-06 by tablatronix

davetcc, 2020-02-06:
Yep thanks I’ll make sure the docs are cleaned up. Good news that it is working for you!

tablatronix, 2020-04-09:
hmm getting these crashes again now with outputs, no idea why, i will have decode this exception

```C++
  switches.initialiseInterrupt(ioFrom8574(0x20, interruptPin), true);
  ioDevicePinMode(switches.getIoAbstraction(), 5,   OUTPUT);
  ioDeviceDigitalWriteS(switches.getIoAbstraction(), 5, true);
```

```PHP
Exception (28):
epc1=0x4021af03 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 01a0
3fffff90:  3fffdad0 00000000 3ffef0c0 40203094  
3fffffa0:  feefeffe 00000000 3ffef1ac 4020bf9c  
3fffffb0:  feefeffe feefeffe 3ffe8544 401014a9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Startup..
```