TcMenu/IoAbstraction #77: Wire error checking handling

tablatronix, 2020-02-09:
Using pcf8574

I was trying to find out how to check the that the i2c device was present (ack) and am not finding any error flags, I do see that 
`bool PCF8574IoAbstraction::runLoop(){` returns status of `endTransmission`

But If i am using  `taskManager.runLoop();` that does not help me

Ideas?

davetcc, 2020-02-09:
The two runLoops are completely separate. It’s a bit unfortunate they both share the same name. Basically think of the pcf8574 run loop as sync. It’s when the required communication is done. So basically you set any pin states, sync and then read any states from read pins. Hope this clears it up.

tablatronix, 2020-02-10:
Right, But I was just thinking If I am using switches runloop for eg. rotary encoder, which is scheduled via taskmanager runloop, then I cannot get to the wire return results to check for errors.

eg. flow

(void)taskManager.runLoop();
(bool)switches.runloop() -> (bool)ioDeviceSync(ioDevice); 
(bool)pcf8574.runloop() -> (bool)wire.endtransmission()

I cannot check writeok etc..

I can maybe call it directly before taskman, but without `needsWrite` satisfied, it doesn't do anything.
```
  IoAbstractionRef iodev = switches.getIoAbstraction();
  bool res = iodev->runLoop();
```

So my thoughts were tracking the last result status maybe in the lib, and exposing a getter.
Does that make any sense?

davetcc, 2020-02-10:
IoDeviceSync is basically the same as calling runloop on the device abstraction, there is no need for both as above. I wish I had called the device abstraction synchronisation method sync. But time has passed and that is not possible. 

But I agree a method to get the last transmission status on switches would be useful.

davetcc, 2020-02-10:
Add a getter to switches that stores the last update ioabstraction status.

tablatronix, 2020-02-10:
Heh not sure if we are on the same page, the only thing I am trying to do is get the result of ‘twiwriteto’ for I2c error handling.

davetcc, 2020-02-10:
I think we are. Just to confirm: Switches will store the Boolean returned by the last sync operation, it will be accessible via a new getter method that will be added.

tablatronix, 2020-02-10:
I think that works, Only problem I see if that the status would only really exist for i2c devices, so could cause confusion for hardware pins users.

Also there are several results , but no enum for them in wire they are hardcoded. But they should be consistent, Ill leave it up to you if you want to expose these , or just a bool `=0` for success.

`unsigned char twi_writeTo`

__AVR__
```
 * Output   0 .. success
 *          1 .. length to long for buffer
 *          2 .. address send, NACK received
 *          3 .. data send, NACK received
 *          4 .. other twi error (lost bus arbitration, bus error, ..)
```

__ESP8266__
```
return 0;  //  success
return 2; //received NACK on transmit of address
 return 3;//received NACK on transmit of data
return 4;//line busy
```

tablatronix, 2020-02-10:
And to quickly explain my reasoning and alot of libraries do this, If your i2c device is addressed wrong or even not connected at all, the library will continue on happily returning bits you ask for from its internal var. You might think you are getting real data. Same if it drops off or locks up.

davetcc, 2020-02-10:
Ack completely understand. I will give it some thought before implementing.

davetcc, 2020-02-16:
At the moment without a refactor risking users, the best the library can do is return a boolean. Success being true. At this point, if the value is false, you could do another read of the device yourself to determine the proper error. Will be in 1.4.11 today

closed 2020-02-16 by davetcc