TcMenu/IoAbstraction #79: Enhancement: arduino io wrappers

tablatronix, 2020-02-13:
How could this library incorporate compatibility with arduino io wrapper helpers. 

Such as 
pulsein
pulseinlong
shiftin
shiftout

and even tone? probably not its a pwm could be done with shift registers fast enough.

davetcc, 2020-02-13:
You can use any Arduino pins that are not already in use independently. Shift in and out are not really needed. There’s a relatively unused IOAbstraction implementation for shift registers. It supports chained shift registers connected to Arduino pins, the shift output has been tested more than the input abstraction. Both make shift registers look like pins. There’s an example packaged for it too.

davetcc, 2020-02-16:
I'm going to close this as you can use any of these functions directly, with any pin you're not already using the pin within the IO abstraction library.

* pulsein - use directly
* pulseinlong - use directly
* shiftin and shiftout: Either use directly or use the IoAbstraction implementation for shift registers. Documented here: https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/ioabstraction-pins-io-expanders-shiftreg/#using-with-shift-registers

Hope this helps, please re-open if you think there's something else I've missed.

closed 2020-02-16 by davetcc