TcMenu/IoAbstraction #80: taskmanager library standalone

tablatronix, 2020-03-25:
Is task manager available as a separate library?
Have you considered making it, or is it derivative of one already out there?

davetcc, 2020-03-27:
I've not really considered that so far, it's not a derivative, it's a fresh piece of work.

What I have considered is to have a flag that removes all the global variables, so that they have to be defined manually. This way there would be no memory footprint involved in having switches embedded.

However, now that all modern Arduino IDEs support nested libraries it is much easier to do this. Would the removal of the globals help, or is it a different reason that brings you to this?

tablatronix, 2020-03-27:
I just like it and would use it as a standalone and are going to. Might be deserving of widespread availability. There are a few out there but they are not all very good.

davetcc, 2020-03-27:
Agree, it probably makes sense to have it isolated from the switch library. It’s certainly a lot easier to do it now that Arduino ide supports dependencies. 

I just need to get tcMenu 1.4 out and then I’ll do it.

Task manager is quite standalone anyway so should be easy to move out.

davetcc, 2020-03-28:
Move the task manager class and it's associated helpers and tests out to a new project. We'll request that the project be included in the Arduino core library set. Once this is actioned the classes will be removed from this package and instead, this library will depend on the new taskManager library.

To be done as soon as tcMenu 1.4 is released.

tablatronix, 2020-03-28:
Add platformio library file also

tablatronix, 2020-06-12:
Is there a way to change the **when** time interval for an existing timer, without recreating it ?

davetcc, 2020-06-29:
Needs to be moved to next release, to de-risk 1.5.0 which is already far too big, moving to 1.5.1.

davetcc, 2020-07-06:
1.5.1 is a patch release with only minor changes.

davetcc, 2020-07-28:
> Is there a way to change the when time interval for an existing timer, without recreating it ?

We now have an internal development version of task manager separated out from IoAbstraction. We'll be putting it on github soon. This should be a fairly easy change, I'll see if it's possible to fit it in.

In TaskManagerIO V1.0 there will also be support for"events". Where you register an event class with taskmanager, and it checks in with your event to see if it should be run, and then calls exec when it should run, the event can then decide if it is finished or can be re-triggered. You can set the granularity of the next call to the event. In the next version, it will also be safe to call the add methods even on another thread.

closed 2020-09-01 by davetcc