TcMenu/IoAbstraction #81: Question: how to setAccelerationMode on RotaryEncoder

tablatronix, 2020-04-03:
Maybe I am missing something obvious..

How do I reference the hardwareRotaryEncoder method, setAccelerationMode if I have a rotaryencoder obj and used setupRotaryEncoderWithInterrupt() to init it..

This gets me the rotaryEncoder
```C++
  setupRotaryEncoderWithInterrupt(encoderAPin, encoderBPin, onEncoderChange);
  switches.getEncoder() // 
```
Alternative workaround is

```C++
  HardwareRotaryEncoder* firstEncoder = new HardwareRotaryEncoder(encoderAPin, encoderBPin, onEncoderChange);
  switches.setEncoder(0, firstEncoder);
  firstEncoder->setAccelerationMode(HWACCEL_SLOWER); // acceleration mode
```

But then I need to keep my firstEncoder pointer around etc..

So My question is how can I get to hardwareRotaryEncoder from rotaryEncoder eg.(switches.getEncoder())

tablatronix, 2020-04-03:
I attempted to fix this, no idea what I am doing though, but it does let me do this, soo.

`switches.getEncoder()->setAccelerationMode(HWACCEL_NONE); `

davetcc, 2020-04-03:
Yep that’s what I was going to suggest. Those helper methods just set the first encoder for you automatically.

tablatronix, 2020-04-03:
Ok let me know if I need to fix that PR, or if I set the virtual funcs properly..

davetcc, 2020-04-06:
Sorry let me take a look at that PR today.

davetcc, 2020-04-06:
To get a HardwareRotaryEncoder from `switches.getEncoder()` you'd just do either a C or C++ cast the return of getEncoder() to the right type. This is because in your case you've put a hardware encoder into slot 0, so you can safely do the case below. If you had put another type of encoder in that slot, the below would fail. See the class structure: https://www.thecoderscorner.com/ref-docs/ioabstraction/html/class_rotary_encoder.html

In summary, in code, you'd just do:

`HardwareRotaryEncoder* hwEncoder = reinterpret_cast<HardwareRotaryEncoder*>(switches.getEncoder());`

Or if you like old school C:

`HardwareRotaryEncoder* hwEncoder = (HardwareRotaryEncoder*)switches.getEncoder();`

Thanks,
Dave.

tablatronix, 2020-04-06:
doh, I had no idea you could do that, I thought I read that but didn't understand it.
Works wonderfully, Thanks for your time.

```C++
void setEncoderAccel(HWAccelerationMode accel){
  HardwareRotaryEncoder* hwEncoder = reinterpret_cast<HardwareRotaryEncoder*>(switches.getEncoder());
  hwEncoder->setAccelerationMode(accel);
}
```

closed 2020-04-06 by tablatronix