TcMenu/IoAbstraction #83: HardwareRotaryEncoder fails silently if changeprecision not set

tablatronix, 2020-04-13:
This is an odd design choice or a bug and has caught me up over and over..

Rotary encoder does not support callbacks for simple changes, it only allows callbacks if
`maximumValue` is set to non 0 via changePrecision, which is confusing as it is not required as a parm or constructor.

`void RotaryEncoder::increment(int8_t incVal) {`

Does nothing if maxvalue is 0, see my code comments below
I think it would be nice to fix this to avoid this error (default maximumValue in constructor), or also change the behavior and allow a simple callback for change, since that is all most people need to know, how many steps and which direction..

```C++
void RotaryEncoder::increment(int8_t incVal) {
	if(incVal >= 0) {
		if(currentReading + incVal <= maximumValue) {
			currentReading = min((uint16_t)(currentReading + incVal), maximumValue);
			callback(currentReading);
		}
		else Serial.println("NO VALUE SET +1" + (String)incVal);
	}
	else if(currentReading != 0 && currentReading < abs(incVal)) {
		currentReading = 0;
		callback(currentReading);
	}
	else if(currentReading != 0) {
		currentReading += incVal; 
		callback(currentReading);
	}
	else Serial.println("NO VALUE SET -1" + (String)incVal);
```

Maybe just change this to return `incVal` if max not set ?

tablatronix, 2020-04-14:
Also this functions more like a clamp limits than anything to do with "precision"

davetcc, 2020-04-14:
Agreed on the naming, it should really have been setRangeAndValue, I may add that method and just make it call the underlying changePrecision existing method. On the zero setting front, I'll need to take a look through the implementation to see what's possible in this case. This would also suit another user trying to use the encoder in an infinite way.

davetcc, 2020-04-15:
See #85, it is another request for this functionality.

tenmilez, 2020-04-15:
In my opinion, at the most basic level, this library should callback with the direction of rotation for every click that happens. I see this as returning a +1 or -1 depending on direction. 

As a practical matter, I could see a situation where if you rotate quickly, generating interrupts faster than the callback can complete, that it loses track somehow. In this case it would be helpful to have a callback with +/- the number of rotations, but you'd have to specify some kind of debounce* threshold. 

Then you could layer things like value tracking, velocity, etc on top of those functions. 

*I'm referring to a kind of debouncing that's a little different than the debouncing normally associated with hardware like rotary encoders. At some level you still need to debounce individual clicks, but in this case, to determine total number of clicks to be returned for a callback, you'd need to identify how long after the last input to wait before returning the number of clicks.

tablatronix, 2020-04-15:
That is what the library I am replacing does, and I am simply using my own counters with this library now to infer the deltas, except I have to reset the encoder values when it gets to its limits also or it just stops firing callbacks, that is my current work around for this issue.

davetcc, 2020-04-15:
I think there are at least two modes of operation for rotary encoders:

1/ used to scroll between 0 and some value, to represent something in the real world, like a position in a menu, volume in an amplifier, range of values in an enumeration.

2/ used to find the direction of an event that can go on forever in one direction. In this case, the direction and acceleration information need to be captured. In an ideal world it would count up the number of "missed" ticks since the last signal, but this may need to come in a later pass. In the initial pass, maybe it just returns +1 or -1 and disables hardware acceleration detection.

Once step 2 is implemented I think a second issue should then be raised to break the rotary encoder support into two parts as you describe. So that anyone who wants to use it as per 2 above uses the HardwareEncoder core class.

davetcc, 2020-05-12:
Direction only mode is now on master and scheduled for the next release. Just set the maximum to 0. Note that it does not store up ticks presently. It would be good if you could take a look and close if it works for you.

davetcc, 2020-05-13:
Closing out the issue now, there's the question around keeping track of numbers mentioned earlier, but I think at the moment it may be quite a big change, please feel free to raise another issue for that if it's really important and I'll need to try and fit it into a release somewhere. Especially if you can see an easy way to fit it into the current structure that I'm missing.

tablatronix, 2020-05-13:
Ill test this today

tablatronix, 2020-05-13:
Works great!