TcMenu/IoAbstraction #84: i2c Dacs abstraction Discussion

tablatronix, 2020-04-14:
I am using a bunch of MCP4725 i2c DACs, do you think the new analog abstraction capability would be ideal for using in that instance ? ( I DO see it is only pins atm ) I am curious...

davetcc, 2020-04-14:
Yeah, this is the long term aim - I have grand plans for it including a data streaming API. But honestly, that's probably another library, for another day.

davetcc, 2020-07-25:
Actually, I'd like your feedback on the current analog interface. I've acquired a few more analog devices, DACs, digital potentiometers, and a high-end burr brown volume control. I'm thinking that I'll create a sub-project IoAbstractionAnalog that has the I2C, I2S and SPI analog abstractions.

This additional library will only be aimed at bigger 32 bit devices, so memory footprint will not be considered. To start with I'll add a few i2c and spi devices and make it really easy for people to add other ones in.

If you have any thoughts on this it would be great, as my ideas on it are still a bit sketchy. I have a few projects coming up that need the analog abstraction more complete than it is now.

davetcc, 2023-02-15:
First wave, SPI and I2C digital potentiometers and SPI volume controls

davetcc, 2023-05-21:
I'm going to close this for now and open separate issues as I (or others) implement each device over time, the first one is implemented, which is a PGA231x that sets the left and right volume as channels 0 and 1.

closed 2023-05-21 by davetcc