TcMenu/IoAbstraction #85: How to use a rotary encoder as a rotary encoder and not a potentiometer

tenmilez, 2020-04-15:
In your example for a rotary encoder we get to define `void onEncoderChange(int newValue) ` and we have to specify a max value elsewhere. It would be far easier to get a potentiometer for this purpose. 

I am using a rotary encoder specifically for the purpose of determining direction of rotation and not having a maximum or minimum value. Can IoAbstraction support that?

davetcc, 2020-04-15:
Please take a look at the comments in issue https://github.com/davetcc/IoAbstraction/issues/83. We are thinking of making it such that if the maximum value is set to 0, it just returns the increment value. IE: the direction and acceleration information.

Let me know if this would work for you. It's scheduled for the next release.

davetcc, 2020-05-12:
Direction only mode is now on master and scheduled for the next release. Just set the maximum to 0. Note that it does not store up ticks presently. It would be good if you could take a look and close if it works for you.

davetcc, 2020-05-13:
Will be released in 1.5.0. Closing out issue.

closed 2020-05-13 by davetcc