TcMenu/IoAbstraction #86: ESP8226 Compatibility

frantek, 2020-05-14:
ISR's on ESP8226 need ICACHE_RAM_ATTR attribute declared. I suggest a change such as this ti SwitchInput.cpp

`#if defined(ESP8266)
void ICACHE_RAM_ATTR registerInterrupt(uint8_t pin) {
#else
void registerInterrupt(uint8_t pin) {
#endif
	taskManager.setInterruptCallback(onSwitchesInterrupt);
	taskManager.addInterrupt(switches.getIoAbstraction(), pin, CHANGE);
}
`

davetcc, 2020-05-15:
Many thanks, we'll get this into 1.5.0. We had a cycle of getting this working further down the stack, for rotary encoders, but I think this one must have been missed.

davetcc, 2020-06-19:
As far as I can tell, code that needs the ram cache attribute is in task manager, because it is the only code that runs in a raw interrupt handler. Everything else is marshaled by task manager, so I *at least think* that switches / SwitchInput should be safe.

However, I'm more than open to anything that I've missed here, as writing interrupt / threaded code is very complex and thread-safety difficult to achieve.

Have you seen any issues caused by missing cache attributes? If so, please can you share the crash dumps so that I can determine where the attribute was missing.

davetcc, 2020-06-29:
removing milestone because it's not clear if there has been a crash, or if a change needs to be made, the suggestion in the initial comment is not run in an ISR. Please provide a crash report if you have one available.

davetcc, 2020-07-11:
As far as we can tell, the code highlighted does not need to be marked with ISR_ATTR, registerInterrupt is called within user-level code to register the interrupt handler. The only code called by an ISR is in taskManager, and I think everything needed is marked ISR_ATTR. Please re-open if you disagree and we'll discuss.

closed 2020-07-11 by davetcc