TcMenu/IoAbstraction #89: Taskmanager exception esp8266

tablatronix, 2020-06-14:
unfortunately my stack trace is pretty useless..

But its a soft wdt reset so it could be a loop

```php

0x4020f37b: TimerTask::microsFromNow() at ??:?
0x4020f460: TaskManager::putItemIntoQueue(TimerTask*) at ??:?
0x4020f665: TaskManager::runLoop(bool) at ??:?
0x40206f86: loop at ??:?
0x40213968: loop_wrapper() at core_esp8266_main.cpp:?
0x40101779: cont_wrapper at ??:?
```

Possibly this loop, add a delay(0)

```C++
	while (current != NULL) {
		if (current->microsFromNow() > tm->microsFromNow()) {
			previous->setNext(tm);
			tm->setNext(current);
			return;
		}
		previous = current;
		current = current->getNext();
	}
```

I will see if I can reproduce and eliminate it.
Obviously all loops should have a delay on esp8266, but timing critical stuff might need special handling or no interrupts also..

davetcc, 2020-06-14:
This looks like a bug in the way task manager is adding to the linked list. That loop should exit within a couple of iterations.  If it’s reproducible please send me or attach a sketch, so that I can track it down. It looks like task manager is stuck. There’s a lot of tests around this but there’s always edge cases.

davetcc, 2020-06-29:
I think I've come up with an edge case that would cause the watchdog to trigger. It is when using yieldForMicros in longer delays, basically, it could cause the watchdog to fire before the function completes, as it only calls yield() once at the start. I've now made it so that yield() is called after every task. Hopefully this will go some way to fixing up longer running tasks, and longer yieldForMicros calls.

If you manage to recreate the crash let me know, I'd be glad to have a sketch that can reproduce.

tablatronix, 2020-06-30:
I changed some code and never saw it again. I will try to reproduce, I think I added some sanity checking to avoid creating the same timer over and over or something like that. Ill rollback and see if I can isolate it.

davetcc, 2020-07-11:
Please re-open if you can recreate the issue so that it can be investigated. I think what you saw was just taskmanager in a yield for too long, so it may be fixed in 1.5.

closed 2020-07-11 by davetcc