TcMenu/IoAbstraction #91: wire.h eeprom abstraction error

tablatronix, 2020-07-02:
2 problems, EepromAbstractionWire.h is being loaded even if not used
and I also get wire.h error, 

this is the buttontoraryencoder example

```
In file included from /Users/shawn/projects/microcontrollers/dev/libraries/IoAbstraction/src/EepromAbstractionWire.cpp:6:0:
/Users/shawn/projects/microcontrollers/dev/libraries/IoAbstraction/src/EepromAbstractionWire.h:18:18: fatal error: Wire.h: No such file or directory`
```

davetcc, 2020-07-02:
I’ll take a look at this tomorrow and see what’s happened there. What Was the target board?

tablatronix, 2020-07-02:
Nodemcu v1

davetcc, 2020-07-03:
I see what's happened here I think now, I'm assuming you're using platformIO as you've discussed that before here.

I had a definition in my platformIO env to bring in Wire as a library dependency, as I'd seen that in other projects. See my nodeMcu definition below. We test IoAbstraction fairly heavily on ESP8266 using a Node MCU. 

```[env:NodeMCU]
upload_protocol = esptool
platform = espressif8266
framework = arduino
board = nodemcuv2
monitor_speed = 115200
build_flags = -DIO_LOGGING_DEBUG=1
lib_deps = Wire, u8g2```

Do most users rely on the header file scanning for libraries? If so, I'll remove the Wire dependency and do a patch release.

As a temporary measure, you can add Wire and it compiles fine and runs through all the test cases.

davetcc, 2020-07-03:
BTW, in terms of EeepromAbstraction being brought in, it's the way that Arduino code compiles, any cpp files in the libraries source directory are compiled regardless of usage. It's the linker that determines what stays and what goes. So if you're not using EepromAbstraction, it probably won't be there at runtime, but it will be compiled.

Wire is a real problem because once it's brought in it has a major effect on smaller boards memory.

IoAbstraction being so large dates back to when Arduino IDE didn't have dependency management, written today it would have started out as three libraries. That's what we are going to do next, I just need to get tcMenu 1.4 out first.

davetcc, 2020-07-03:
More information after reading through https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

I'm assuming that you're using chain mode for library dependency finder, because that would have worked before and would now fail because it is not evaluating #ifdef expressions, if you were using that chain+ mode evaluates the ifdefs in the code.

It appears that Arduino IDE seems to handle this better, as all the projects we tried appeared to work there.

tablatronix, 2020-07-03:
ah ok, Yeah I have to add wire.h to get examples to compile , which is annoying for newbs, to have examples broken they will immediately go install another library.

davetcc, 2020-07-03:
Agreed on the non-compilation front in PIO, nothing worse than encountering that. However, I think I've got a fix for it, I just need to test it myself first and then I'll make a patch release. My plan is to add the Wire dependency directly in the library.json file when the platform is Arduino, therefore it will be there by default.

In terms of frameworks, both mbed and Arduino basically compile everything and then let the linker exclude things that are not needed. The EEPROM classes should not be in the output however if not used, but it's a bit difficult to get your head around at first.

Apologies that you've hit this, we went through a huge testing cycle before releasing 1.5.0 and thought we covered everything. But I agree this behaviour is not right.

davetcc, 2020-07-06:
I think the easiest thing to do, is to include wire in each ino file, with a comment indicating why it's needed. I think the vast majority of existing users will have Wire.h included already, and for the few that don't, I'll put an entry on theCodersCorner forum and put something on the front page of the repo for platformIO users.

This seems to be a significant difference between how platformIO and Arduino work. Arduino seems to somehow hook into the compiler, using its syntax tree to work out when something is included. Whereas platform IO uses a python script to determine which includes are in scope, and the python script needs to be in "chain+" mode to work like Arduino IDE.

So, all examples should compile out of the box, and platformIO users will either have to include wire.h in their project header / INO, or use chain+ mode. I think this is the simplest solution that requires the least "magic". But I'm open to suggestions on it to be honest.

davetcc, 2020-07-09:
Actually, in the end, after even more reading up on platformIO dependency manager I found that it has support for platform-specific dependency listing in library.json, so I added Wire to the list. I went through many of the examples in platformIO and Arduino and they all seem to work with this change. It removes the need to include Wire.h (unless you're actually using it)

closed 2020-07-09 by davetcc