TcMenu/IoAbstraction #95: Switch input callbacks not being invoked

MonitorComputing, 2020-07-31:
Using a pair of PCF8574s for eight pushbutton switch inputs and eight LED ouputs. Each pushbutton is used to toggle a corresponding LED.

Buttons one through to five work as expected. Buttons 6 through to eight are not causing their LEDs to toggle.

Have a press and release callback for each button input. To debug added a scan all inputs function which is invoked at the start of either callback function. This shows that if any of buttons six through to eight are held down then any of buttons one through to five are pressed and released the callbacks for whichever of buttons one through to five was pressed are invoked, but not for buttons six through to eight, and in those callbacks the input scan shows that whichever of buttons six through to eight have been pressed and held is seen as active.

Hence it would seem that buttons six through to eight are being read correctly but that the callbacks registered for them are not being invoked.

Unfortunately I do not have direct access to the hardware being used nor any other facility to directly debug the execution myself. I've made a cursory examination of the IoAbstraction library source but did not see any reason for the errant behaviour.

I've attached the Arduino sketch along with screen captures of the serial console output during testing.
[pcf8574.txt](https://github.com/davetcc/IoAbstraction/files/5006606/pcf8574.txt)
[image]
[image]
[image]

davetcc, 2020-08-04:
Apologies for the delay in getting back.. Please could you tell me which processor type you are targeting and also the version of IoAbstraction (from library.properties in the IoAbstraction dir).

Before version 1.5 switches did not reallocate automatically and therefore hit a hard limit unless you redefined the value in SwitchInput.h

For some architectures it was suspiciously set at 6!

This is no longer required with 1.5 and it should reallocate automatically. Can you confirm versions first, and we'll take it from there.

MonitorComputing, 2020-08-04:
I raised the issue on behalf of someone else so there'll be a slight delay whilst I get that information from them. I'll post back as soon as I have the answers.