TcMenu/IoAbstraction #97: Write on multiple Shift Register do not work

giancarlodomanico, 2020-09-04:
Writing on two or more chained shift register do not work.
I believe there is a mistake in the function ShiftRegisterIoAbstraction::writePort.
Before exiting the function the computed value of 'toWrite' is replaced with the int8 value that have to be set on the port that the pin belongs to.
The line 

toWrite = portVal;    

should be removed. 
I have tested the change and in this way it works fine for me.

void ShiftRegisterIoAbstraction::writePort(uint8_t pin, uint8_t portV) {
	uint32_t portVal = portV;
	if(pin < SHIFT_REGISTER_OUTPUT_CUTOVER) return;
	pin = pin - SHIFT_REGISTER_OUTPUT_CUTOVER;
	if(pin < 8) {
		toWrite &= 0xffffff00;
		toWrite |= portVal;
	}
	else if(pin < 16) {
		toWrite &= 0xffff00ff;
		toWrite |= (portVal << 8);
	}
	else if(pin < 24) {
		toWrite &= 0xff00ffff;
		toWrite |= (portVal << 16);
	}
	else {
		toWrite &= 0x00ffffff;
		toWrite |= (portVal << 24);
	}
	//toWrite = portVal;    // should be removed
	needsWrite = true;
}

davetcc, 2020-09-07:
It would be great if you could test the master version committed just now, it should fix the issue. If it does, I'll do a patch release soon after.

giancarlodomanico, 2020-09-07:
Sorry Dave but I not see any fix in method ShiftRegisterIoAbstraction::writePort. You will overwrite the value set in toWrite

giancarlodomanico, 2020-09-07:
If I'm not wrong you have committed to ioa-taskmgr-refactor and not to master. Is this the version you want I test ?

davetcc, 2020-09-08:
Ah sorry, I will redo that now, my IDE got stuck on an old branch :(

davetcc, 2020-09-08:
Apologies for that, it's now on master!

davetcc, 2020-09-14:
Ensured that this will not affect compilation and now closing. Fix is as per previously tested configuration

closed 2020-09-14 by davetcc