TcMenu/IoAbstraction #102: Add hwacceleration as parameter in rotary encoder setup function

markus-i, 2020-11-19:
I would have opened a pull request (since I wouldn't consider this a large issue), but don't know how. What I did was basically add the HWAcceleration as parameter with default setting to the setupRotaryEncoderWithInterrupt function. Everything else remains unchanged.
Only tested for a single encoder on Arduino.

[SwitchInput.zip](https://github.com/davetcc/IoAbstraction/files/5567751/SwitchInput.zip)

Reason for the change: I think it unlikely that someone wanted to change the acceleration setting during runtime (but you should still be able to do so) and see that setting rather as a property of the device. I find this easier than getting the pointer to the HardwareRotaryEncoder and then calling the set() function.

Removing the set() function _might_ even save a few bytes, since the compiler/linker _should_ then recognize the attribute as compile-time and not run-time. But I didn't want to remove existing functionality.

davetcc, 2020-11-23:
Many thanks for providing this, I'll try and get it merged in shortly.

davetcc, 2020-11-28:
Added to 1.6.5 release target.

davetcc, 2020-11-29:
I've made the change on master, I made the final parameter on HardwareRotaryEncoder to ensure backward compatibility for users with more than one encoder.

Please could you check if it does what you wanted on master?

markus-i, 2020-11-29:
Short check - works for me. Thanks!

closed 2020-12-06 by davetcc