TcMenu/IoAbstraction #110: error reading 74hc165

giancarlodomanico, 2021-02-06:
Hi, I'm using the library to read from multiple 74hc165 and I have seen that the library miss a the first bit.
Testing with my readlilng code it works fine, testing with shiftIn and your code it doesn't work. Looking around the web I see a discussion that explain that this chip need a clock HIGH/LOW different from the one inside shiftIn. 
It is suggested to use this code instead of builtin shiftIn(). It could be fine to have a switch to use builtin shiftIn or the other based on which Shift Register will be used. I have already tested the new function and it works

uint8_t shiftIn165(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder) {
  uint8_t value = 0;
  uint8_t i;

  for (i = 0; i < 8; ++i) {
    digitalWrite(clockPin, LOW);
    if (bitOrder == LSBFIRST) {
      value |= digitalRead(dataPin) << i;
    }
    else {
      value |= digitalRead(dataPin) << (7 - i);
    }
    digitalWrite(clockPin, HIGH);
  }
  return value;
}

davetcc, 2021-02-06:
Thanks for letting me know it works. I’ll move over to this new code for input. A couple of people have said that shift in doesn’t always work now.

giancarlodomanico, 2021-02-07:
Hi I don't know about standard shiftIn because I only have 74hc165 shift registers and they need clock LOW before reading. The standard ShiftIn works also with this chip but of course it will loose one bit. I don't know which other shift registers need clock LOW.
I have added a new class that use the shiftIn165 becuase a switch during the read will waste time.

davetcc, 2021-02-09:
@giancarlodomanico would you please test the master src/IoAbstraction.cpp/.h with a 74HC165 and let me know if it works for you. If it does I'll release it soon after.

There is a new helper method to construct the Io device:

```IoAbstractionRef inputFrom74HC165ShiftRegister(pinid_t readClkPin, pinid_t dataPin, pinid_t latchPin, pinid_t numOfDevices = 1);```

And a new class that provides the functionality, I've cut the shift code back to the minimum, but don't have a device to test it with to hand.

```ShiftRegisterIoAbstraction165In```

davetcc, 2021-02-09:
If you need input and output at the same time use a MultiIoAbstraction, see https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/arduino-pins-and-io-expanders-same-time/

giancarlodomanico, 2021-02-09:
Hi, I have tested your code and it works fine on 74hc165. I have tested it with 2 74hc165 in series and it works fine too

davetcc, 2021-02-09:
Thanks for letting me know. I’ll release it hopefully tomorrow.

davetcc, 2021-02-13:
Tested, closing the issue in preparation for release.

closed 2021-02-13 by davetcc