TcMenu/IoAbstraction #111: Doesn't compile under mbed 6.5 in platformio

flamedmg, 2021-02-17:
I tried to use your great library in platformio using mbed framework. Just created solution with libraries added for tcMenu and ioAbstraction (latest stable versions) does not compile:
```Compiling .pio\build\bluepill_f103c8\libd0a\IoAbstraction\IoAbstraction.o
In file included from .pio\libdeps\bluepill_f103c8\IoAbstraction\src\AnalogDeviceAbstraction.cpp:6:
.pio\libdeps\bluepill_f103c8\IoAbstraction\src\AnalogDeviceAbstraction.h:95:9: error: 'AnalogOut' does not name a type
   95 |         AnalogOut* out;
      |         ^~~~~~~~~
Archiving .pio\build\bluepill_f103c8\libc5e\libTaskManagerIO.a
.pio\libdeps\bluepill_f103c8\IoAbstraction\src\AnalogDeviceAbstraction.h: In constructor 'AnalogPinReference::AnalogPinReference(pinid_t, AnalogDirection)':
.pio\libdeps\bluepill_f103c8\IoAbstraction\src\AnalogDeviceAbstraction.h:119:27: error: 'union AnalogPinReference::AnalogPinReferences' has no member named 'out'
  119 |                 analogRef.out = new AnalogOut((PinName)pin);
      |                           ^~~
.pio\libdeps\bluepill_f103c8\IoAbstraction\src\AnalogDeviceAbstraction.h:119:37: error: expected type-specifier before 'AnalogOut'
  119 |                 analogRef.out = new AnalogOut((PinName)pin);
      |                                     ^~~~~~~~~
```

I followed the code and found that AnalogOut is not defined due to  DEVICE_ANALOGOUT is not defined and whole AnalogOut.h is excluded from the build. This seems to be ok as board does not support DAC or there may be a board that doesn't support it. I actually can't find where DEVICE_ANALOGOUT  and DEVICE_ANALOGIN are configured for my board. But anyway, my hopes where that absence of this functionality not break the library. I tried to use same library using arduino framework and it works just fine there.

davetcc, 2021-02-17:
We can probably easily work around that one if there’s a clear define we can pick up on. It will be In the AnalogAbstraction I suspect.

davetcc, 2021-02-17:
In https://github.com/davetcc/IoAbstraction/blob/master/src/AnalogDeviceAbstraction.h we need to protect AnalogOut code (three places I can see) when DEVICE_ANALOGOUT (check define is correct) is not set. I don't think there are any boards that do not have AnalogInput or PWM.

davetcc, 2021-02-17:
There was to say the least a series of problems in AnalogDeviceAbstraction that should never have seen the light of day!

However, it is fixed now and although I don't have that device, I tried purposely switching that flag and it seems to compile

flamedmg, 2021-02-17:
I'm ready to help you to test. Should i try to do that in platformio or mbed studio? Should i use master branch of libs to test?
I used this to test:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = mbed
upload_flags = -c set CPUTAPID 0x2ba01477
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT
debug_init_cmds =
  set CPUTAPID 0x2ba01477
  target extended-remote $DEBUG_PORT
  $INIT_BREAK
  monitor reset halt
  $LOAD_CMDS
  monitor init
  monitor reset halt
lib_deps =
     https://github.com/davetcc/TaskManagerIO#master
     https://github.com/davetcc/IoAbstraction#master
     https://github.com/davetcc/tcMenuLib#master
```
The error in the topic went away but solution still not compiling, and now it fails in tcmenulib code:
```
Compiling .pio\build\bluepill_f103c8\lib33c\tcMenu\MenuItems.o
In file included from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/BasicIoAbstraction.h:21,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/IoAbstraction.h:18,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.h:18,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.cpp:7:
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/SimpleCollections.h:90:33: warning: implicitly-declared 'constexpr KeyboardItem& KeyboardItem::operator=(const KeyboardItem&)' is deprecated [-Wdeprecated-copy]
   90 |         binTree[insertionPoint] = item;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.cpp:52:1: note: because 'KeyboardItem' has user-provided 'KeyboardItem::KeyboardItem(const KeyboardItem&)'
   52 | KeyboardItem::KeyboardItem(const KeyboardItem& other) {
      | ^~~~~~~~~~~~
In file included from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/BasicIoAbstraction.h:21,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/IoAbstraction.h:18,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.h:18,
                 from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.cpp:7:
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/SimpleCollections.h: In instantiation of 'bool BtreeList<K, V>::checkCapacity() [with K = long unsigned int; V = KeyboardItem]':
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/SimpleCollections.h:81:13:   required from 'bool BtreeList<K, V>::add(V) [with K = long unsigned int; V = KeyboardItem]'
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.cpp:160:29:   required from here
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src/SimpleCollections.h:111:15: warning: 'void* memcpy(void*, const void*, size_t)' writing to 
an object of non-trivially copyable type 'class KeyboardItem'; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  111 |         memcpy(replacement, binTree, sizeof(V) * itemsInList);
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.cpp:7:
Archiving .pio\build\bluepill_f103c8\lib5fe\libIoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96.a
.pio\libdeps\bluepill_f103c8\IoAbstraction@src-642d8f9bb6d2d5dfe6196d1931d43b96\src\SwitchInput.h:97:7: note: 'class KeyboardItem' declared here
   97 | class KeyboardItem {
      |       ^~~~~~~~~~~~
In file included from .pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/GfxMenuConfig.h:7,
                 from .pio\libdeps\bluepill_f103c8\tcMenu\src\graphics/BaseGraphicalRenderer.h:15,
                 from .pio\libdeps\bluepill_f103c8\tcMenu\src\BaseDialog.cpp:8:
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:128:14: error: use of enum 'IconType' without previous declaration
  128 |         enum IconType : byte {
      |              ^~~~~~~~
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:128:25: error: 'byte' was not declared in this scope
  128 |         enum IconType : byte {
      |                         ^~~~
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:128:30: error: default member initializer for unnamed bit-field
  128 |         enum IconType : byte {
      |                              ^
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:138:14: error: use of enum 'MemoryLocation' without previous declaration
  138 |         enum MemoryLocation : byte {
      |              ^~~~~~~~~~~~~~
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:138:31: error: 'byte' was not declared in this scope
  138 |         enum MemoryLocation : byte {
      |                               ^~~~
.pio\libdeps\bluepill_f103c8\tcMenu\src/graphics/DrawingPrimitives.h:138:36: error: default member initializer for unnamed bit-field
[... 175 lines cut ...]
[... cut ...]
.pio\libdeps\bluepill_f103c8\TaskManagerIO@src-3f35df5c7fa7cc822fe6f7ba800170bd\src\TaskPlatformDeps.h:23:12: fatal error: rtos.h: No such file or directory

**************************************************************
* Looking for rtos.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:rtos.h"
* Web  > https://platformio.org/lib/search?query=header:rtos.h
*
**************************************************************
```
As i understand it complains that mbed does not import RTOS.
Could you please share any  platformio or mbed studio example that works and compiles for any stm board?

flamedmg, 2021-02-17:
Then i realized that tcmenu master branch is not yet production ready and switched to latest stable by doing this:
```
lib_deps =
     https://github.com/davetcc/TaskManagerIO#master
     https://github.com/davetcc/IoAbstraction#master
     davetcc/tcMenu @ ^1.7.1
```
Not it compiles! But consuming to much resources, please take a look:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  47.2% (used 9664 bytes from 20480 bytes)
Flash: [=======   ]  67.6% (used 44272 bytes from 65536 bytes)
```

This is an empty project, no single line of my own code :)

davetcc, 2021-02-17:
TcMenu compilation problems: you're trying to use tcMenu master which is moved to 2.0 BETA, I would keep tcMenu on the last release version for the moment (or alternatively use the "BETA" stream in tcMenu designer). But be aware we've not tested the beta master on mbed yet. Most development is done these days on ESP8266/32, then we get it working on mbed. It would at least compile if you used the BETA stream of plugins though.

In terms of non-rtos use, see the taskManagerIO issue documenting problems with platformIO and non-rtos builds. We've left it open for now as there is no answer on how to do non-rtos 6 with platformIO. https://github.com/davetcc/TaskManagerIO/issues/17

Board size: to be honest, that board may be too small on mbed for tcMenu. You may be better using that with the Arduino framework, but even then it may be a close call.

davetcc, 2021-02-17:
Ah you must have taken tcMenu 2.0 library in the few days it was published, before being removed from the platformIO repository. It was auto published there when it should not have been. I think that 1.7.1 has gone back to being the default.

flamedmg, 2021-02-17:
Thank you for your advice! 
Actually i was using esp32 for most of my development needs and liked most of it, except bad ADC, and few other problems, like unusable pins, and overal small pin count and i was shifting toward stm32 to replace arduino/esp32 with something more widely used / "professional". I don't like arduino in general, but mostly for the problems your great libraries are tend to solve (ioabstraction / timer), but i can't beat that many good guys shared their libs / drivers for that platform!
Summarizing, what board / platform are you suggesting for hobbist with serious needs ? :)My last project was pellet stove, where avr chip died and i replaced it with esp32 with own created software. Works for several month is my house and i'm perfectly happy with it.

davetcc, 2021-02-17:
My views on boards:

* For mbed RTOS, I use anything from STM32F4 range and upwards.
* For Arduino 32 bit, ESP32 with IoAbstraction to iron out a few issues.
* For 8 bit device MEGA2560.

I have STM32F429 and STM32F439 boards for development. They are quite cheap and very powerful. Have inbuilt debugging, loads of RAM and a large FLASH.

davetcc, 2021-02-17:
Remember that Arduino is no more than a library when used with platformIO build system. You can use what bits of it work, and use alternatives where the implementation of something is not good enough.

If you work with AVR, you can access the avr internals
If you work with ARM, you can access CMSIS
If you work with ESP32, you can access FreeRTOS
Some Arduino boards are actually mbed underneath.

flamedmg, 2021-02-17:
Thank you for your advices! What would be your choice for low power / battery powered solutions ?

davetcc, 2021-02-17:
I've not looked at STM32 boards for battery power, but we did an investigation into an MKR board - Atmel SAMD Cortex M0 units 256K FLASH, 32K RAM - easily runs tcMenu. We even have a low power task manager example for that, it reduces power when it's not busy. You could use Arduino as the core, then use CMSIS for any areas that you are not happy with.

I'm absolutely sure there would be similar STM32 devices for mbed, but we've not looked into it so far.

davetcc, 2021-02-17:
Change tested on mbed6, ESP32, MKR and AVR. Safe to release.

closed 2021-02-17 by davetcc

davetcc, 2021-02-18:
Let me know when you get a suitably sized board for mbed, maybe start a forum post (on our main site) or open an issue in the tcMenuLib repo with the compile errors, the board I am targetting next is STM32F429 with 320x240 screen built into it, but I will need to write a renderer.

flamedmg, 2021-02-18:
I ordered few stm32F411 boards also known as black pills. Will take about a month to arrive here from china. I also ordered few ips displays but not sure if drivers already exists for them, will see.

davetcc, 2021-02-18:
> ips displays but not sure if drivers already exists for them, will see

If you get ILI9341panels, I think you can configure that to work with the onboard ART accelerator - https://www.st.com/resource/en/application_note/dm00287603-lcdtft-display-controller-ltdc-on-stm32-mcus-stmicroelectronics.pdf, mbed display support is nowhere near as complete as Arduino. I usually try and buy a board with the display on to make it easier to wire up and it will have a ready setup ST library in platformio.

I even ended up having to fix up the Adafruit OLED port myself, should you need it, it's in my repo!

flamedmg, 2021-02-18:
I bought last 2 of these: https://www.aliexpress.com/item/32779178430.html?spm=a2g0s.9042311.0.0.1b654c4dzfgRZO
S6D0154X driver, i haven't seen it support by any arduino library, but seller should send some example codes, so at least i will take initialisation from there. Seen very good reviews, should be quality displays for not to much $
I mostly avoided stms because of their price and i'm pretty short in money on my hobby. ili9341 + esp32 are probably 10x times cheaper that nice, but very expensive stm32 board. I will check boards, but may be you have suggestion with display built in ?

davetcc, 2021-02-21:
BTW just to let you know mbed compilation is working again on master now, it was mainly down to the use of byte for the enum type instead of using uint8_t. Moved it all over and also fixed some significant warnings around the btree-list at the same time. When I wrote the collection I made a mistake in how I implemented copying, and I've now fixed that too. 

I've also properly released the mbed Adafruit_GFX fork for OLEDs (eg SH1106 and SSD1306 units): https://github.com/davetcc/Adafruit-GFX-mbed-fork