TcMenu/IoAbstraction #112: Per SwitchInput::addSwitch Add Input Type

JRWR, 2021-02-23:
I'm working in a rather limited IO space on my ESP8266 and I'm trying to interface with SwitchInput but I have some rather oddball logic I need to apply

GPIO0 is always high unless pushed
GPIO15 is high when pushed
GPIO13 needs INPUT_PULLUP

I don't see a good method of adding that single INPUT_PULLUP on GPIO13, Unless I can switch it under SwitchInput and call ioDevicePinMode(arduinoPins, GPIO13, INPUT_PULLUP);

JRWR, 2021-02-23:
This is tested to work currently 

`    switches.initialise(ioUsingArduino(), false);
    pinMode(SelectPin, INPUT_PULLUP);
    switches.addSwitch(LeftPin, onPressed, 10, true);
    switches.addSwitch(RightPin, onPressed, 10, false);
    switches.addSwitch(SelectPin, onPressed, 10, true);
    pinMode(SelectPin, INPUT_PULLUP);`

davetcc, 2021-02-24:
Hi there, yes that's correct, you can override the "pull up" logic on a per switch basis, the other option I sometimes use is to provide an external resistor on that pin, then you can safely leave input pullup in place, it will do nothing from all the tests I've done - in fact you'd need a resistor even the other way around to avoid floating IIRC.

I tend to use ESP32 these days more than ESP8266, but they have a similar issue with a few GPIOs that are input only.

davetcc, 2021-03-01:
@JRWR  did this do what you wanted? Can we close the issue or is there still something outstanding? Thanks.

davetcc, 2021-03-09:
```pinMode(SelectPin, INPUT_PULLUP);```

That should not be needed, the default operation is to use INPUT_PULLUP. when pull-up operation is chosen. AFAIK on Arduino, it is even safe to use when the pin does not have output capability (IE it cannot pull up).

davetcc, 2021-05-12:
Closing for now, please reopen if you think there's anything left to do on this one.

closed 2021-05-12 by davetcc