TcMenu/IoAbstraction #114: ESP32 encoder stops working

tablatronix, 2021-03-28:
I ported a project to ESP32 and I am having an issue with the encoder stops working. I assume its an interrupt issues. Anyone have any ideas what might be casing this or know n issues with esp 32 interrupts breaking or having to be restarted ?

tablatronix, 2021-03-28:
#define ENC_PINA       32
#define ENC_PINB       36
#define ENC_SW         33
switches.initialise(ioUsingArduino(), true);
switches.addSwitch(encoderSWPin, onEncoderSWPressed); // encoder button press
setupRotaryEncoderWithInterrupt(encoderAPin, encoderBPin, onEncoderChange);

I will work on a new sketch and see if I can remove some stuff from my code

tablatronix, 2021-03-28:
Looks like this is a wifi pin interrupt issue I was not familiar with

davetcc, 2021-03-28:
To be honest, we use ESP32 as our main development environment these days, we have ESP32 boards with hardware debuggers for development. Along with STM32F4 imbed boards t's one of our core environments. I just saw you new message, yes there are a few pins that don't work as expected when WiFi is enabled, and there are a few higher GPIO's that don't support pull up too. Those would need external pull up.

davetcc, 2021-03-28:
Did swapping around the pins fix it for you? I think at least some of those chosen pins are input only, IE can't pull up / down.

tablatronix, 2021-03-28:
I have pullups, Its strange.. it just stops, I am running standalone tests now, seems to something else going on.

davetcc, 2021-03-28:
Do you see any error output in the terminal? if you're using platform IO you could turn on the IoAbstraction debugging and see where it stops / or if the board panics.

```build_flags = -DIO_LOGGING_DEBUG=1```

Will output logging to the serial console and may be helpful. Once you run the standalone test, if it is not working then please post it here, and I'll give it a try on a test board tomorrow.

tablatronix, 2021-03-28:
I replaced lib with my own interrupts and it still doing it, so it is nothing specific to ioabs, I am using some timers.
I can close this if you want, seems to be something in platform, gonna debug isr just to understand it

tablatronix, 2021-03-28:
If I disable taskManager runloop it doesn't break.. shrug lol I will keep posting updates

tablatronix, 2021-03-28:
I figured out my issue.. lol
I was calling hallRead() somewhere which connects the hal to S_VP,SVN which iss..... gpio 36,39

Easy to overlook, and not specifically warned about in docs or datasheet.

closed 2021-03-28 by tablatronix