TcMenu/IoAbstraction #120: InputOnly & OutputOnly creation error

giancarlodomanico, 2021-04-18:
Hi,
there is something wrong if I create two reference to IoAbstraction.
This is the code that do not work.

#include <Arduino.h>

#include <IoAbstraction.h>

#define SERIAL_SPEED 115200   

#define READ_DATA_PIN   4       //giallo
#define READ_LATCH_PIN  5       //verde
#define READ_CLOCK_PIN  6       //viola

#define WRITE_DATA_PIN   8      //giallo
#define WRITE_LATCH_PIN  9      //verde
#define WRITE_CLOCK_PIN  10     //viola

#define DATA_WIDTH    8
#define SHIFTIN_NUM_CHIP 2
#define SHIFTOUT_NUM_CHIP 1

IoAbstractionRef inShiftRegister = NULL;
IoAbstractionRef outShiftRegister = NULL;

void setup()
{
  Serial.begin(SERIAL_SPEED);
  
  inShiftRegister = inputFrom74HC165ShiftRegister(READ_CLOCK_PIN, READ_DATA_PIN, READ_LATCH_PIN, SHIFTIN_NUM_CHIP);
  if (inShiftRegister == nullptr)
  {
	Serial.print("Error inShiftRegister");
    delay(1000);
  }  

  outShiftRegister = outputOnlyFromShiftRegister(WRITE_CLOCK_PIN, WRITE_DATA_PIN, WRITE_LATCH_PIN, SHIFTOUT_NUM_CHIP);
  if (outShiftRegister == nullptr)
  {
	Serial.print("Error outShiftRegister");	 
    delay(1000);
  }

}

void loop()
{
}

davetcc, 2021-04-19:
> there is something wrong if I create two reference to IoAbstraction. This is the code that do not work.

I sorry but you'll need to indicate how it does not work. What is the exact issue? 

Does it compile? 
Does it run as expected?

BTW: the way you're using the last parameter of the shift register initialisation is wrong, it's not a chip number, it's how many of that device you've connected together. Unless you have two shift in registers wired in passthrough, SHIFTIN_NUM_CHIP should be 1.

```
#define SHIFTIN_NUM_CHIP 2
#define SHIFTOUT_NUM_CHIP 1
```

giancarlodomanico, 2021-04-19:
Hi Dave,
the issue is that Arduino will not start or it will reset but it is not easy understandable. 
SHIFTIN_NUM_CHIP is right, I have 2 SHIFTIN chip in cascade becuase I need 16 buttons.
It seems like the second IoAbstractionRef has not been created. Is is null but also the test for null will not work.
This is a snippet and not my full code. If you want I can send you also the full code to your private email. I cannot attach it to this issue because it is for a customer and I cannot share it with all

davetcc, 2021-04-19:
This is significant usage of the shift register support. Most people go for MCP23017 of PCF8574 I2C devices. At this point I think the best thing to do is go back to the example - exampleShiftReg, alter it for your board arrangements and try that in isolation, determine if it works with your hardware. Often it's too difficult to debug as part of a complex program. If you have a hardware debugger, I would put a breakpoint at the second IoAbstractionRef and see what happens there. if you don't have a hardware debugger, put a print statement after the 2nd IoAbstractionRef showing the pointer.

When you say it seems like the abstraction has not been created, what makes you think it has not been created?

davetcc, 2021-04-19:
> I will try to find out a better test case to isolate the issue.

I think the only way to proceed here is to modify the shift register example exampleShiftReg and test the input and the output example one at a time. Then combine them both. This is likely some kind of integration issue somewhere.

davetcc, 2021-04-20:
Something to bear in mind when using the helper functions to build a shift register they use the new operator, could it be the board running out of heap? Here's the actual code:

https://github.com/davetcc/IoAbstraction/blob/2a8fd3fcbfbfd5c44d882bb14c6715a306103558/src/IoAbstraction.cpp#L130

and

https://github.com/davetcc/IoAbstraction/blob/2a8fd3fcbfbfd5c44d882bb14c6715a306103558/src/IoAbstraction.cpp#L209

There's nothing that would stop you from trying to create those objects globally and remove the new operator altogether. It would rule out a whole load of problems immediately. For example at global scope:

    ShiftRegisterIoAbstraction165In myShiftReg(readClkPin, dataPin, latchPin, numOfDevices);
    IoAbstractionRef inputRegisterRef = &myShiftReg;

An IoAbstractionRef is nothing more than a pointer to the shift register device. That's what the 2nd line is doing.

giancarlodomanico, 2021-04-25:
Hi Dave,
you are right it has been a heap issue. It sound strange but I have got to remove all Serial.print to let the program run. I have used a  Arduino Nano because the program in the beginning has to be very simple but later it has been grown and the Nano is too small. I will move to a 'bigger' device. Thank you for your help.

closed 2021-04-25 by giancarlodomanico