TcMenu/IoAbstraction #124: undefined reference to `internalDigitalIo()' since update to version 2.0.0

kimble4, 2021-05-12:
I don't have time to look into this properly right now, but since updating to 2.0.0 compile for ESP32 fails with:

```
/tmp/arduino_build_330010/libraries/LiquidCrystalIO/LiquidCrystalIO.cpp.o: In function `LiquidCrystal::init(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, BasicIoAbstraction*)':
/home/kim/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Print.h:80: undefined reference to `internalDigitalIo()'
/tmp/arduino_build_330010/libraries/IoAbstraction/IoAbstractionWire.cpp.o: In function `PCF8574IoAbstraction::attachInterrupt(unsigned char, void (*)(), unsigned char)':
/home/kim/documents/programming/sketchbook/libraries/IoAbstraction/src/IoAbstractionWire.h:146: undefined reference to `internalDigitalIo()'
/home/kim/documents/programming/sketchbook/libraries/IoAbstraction/src/IoAbstractionWire.h:146: undefined reference to `internalDigitalIo()'
```

Works fine if I revert to 1.7.5

davetcc, 2021-05-12:
Thanks for reporting this, we know what it is, and we'll do a small patch now.

davetcc, 2021-05-12:
There was a small error in one of the headers when ESP32 not in EXTRAS mode, this is a new mode we've been testing, but have not yet documented where the IDF functions are used for digital IO. The one test we missed was extras mode off! Apologies and fixed now. We'll just do a few smoke tests on other boards and then release this evening.

davetcc, 2021-05-12:
Tests to run before release:

ESP32 - EXTRAS off  [tested]
ESP32 - EXTRAS on  [tested]

ESP8266 - any I2C sketch
MKR - any I2C sketch
AVR - any I2C sketch

closed 2021-05-12 by davetcc