TcMenu/IoAbstraction #129: ESP32 Interrupt issue?

tablatronix, 2021-06-12:
I just noticed that if I disable ` taskManager.runLoop();` my encoder stops working..
So are interrupts not actually working ?

Is this normal behavior or is something wrong ?

`switches.initialise(ioUsingArduino(), true);`
`setupRotaryEncoderWithInterrupt(encoderAPin, encoderBPin, onEncoderChange);`

davetcc, 2021-06-12:
You can’t really do much in an actual ISR. We raise a flag in the ISR  that gets picked up in the next run loop. This is a very common pattern.

You can have more than one task manager on different threads but you must make sure any shared state is suitably accessed using either atomic constructs or a mutex. 

For example you could move heavy operations onto another task manager and treat it like a queue.

davetcc, 2021-06-13:
Here's a more complete answer to the above, the other was sent from my mobile yesterday.

ISRs are dangerous places, many IDF/Arduino functions that you use could crash when used in an interrupt context. Further, you stop all other processing on that core until the interrupt completes. The best practice for interrupt handling is to do as little as possible and return, and this is how TaskManagerIO handles it. In fact in newer arrangements, we have started moving towards events, but they work the same way, you mark them interrupted, and then the run loop will execute them ASAP.

Further, there is no way that it would be safe to make the callback within an interrupt context, nobody would expect to be in an interrupt callback at that point.

For power saving in the loop method, you can query for how long a delay there is to the next task and go into a low power state that wakes on interrupt. See the low power example on our website for more details.

Admittedly, one big problem on Arduino is that the I2C and SPI (especially I2C as it's slow) are often synchronous causing major headaches and task contention. However, on ESP32 and mbed we are looking to get around the contention by using their asynchronous I2C support, it will need some reworking of our APIs to get there, but nowhere near as much as we originally thought. However, the libraries that you use will be another matter, and each would need to be evaluated for latency in turn.

I'm not familiar with the architecture of your system, but if you have some long-running tasks then yes you could move them onto another task manager. You can create a queue of classes extending `Executable` / `BaseEvent`, or even lambdas with capture (enabling the flag and checking the dynamic memory cost) onto another task manager. You could create immutable objects (reducing the risk of cross-thread visibility problems) and the return value could be declared as an atomic type. The return could be an event posted back onto the originating task-manager or a callback. Be aware if you were to make it a direct callback, you'll be calling that code on another thread context. In this case, be very careful that what you are calling is safe to use in that context. On a dual-core ESP32, volatile may not be enough depending on context.

Just to be clear: Task manager is a lock-free bus design, which means that there is no memory barrier during task queuing. All thread safety is achieved using CAS operations and memory addresses that never change. Interrupts and multi-threading are the hardest things we have to deal with as programmers, often processors and memory don't work as we would initially expect, and the effects of running something on another thread go right through to the underlying system and libraries we call. I've spent nearly 25 years writing threaded code in C++, Java, and C# and it always takes effort to get it right.

tablatronix, 2021-06-13:
I know about ISR, I just want to decouple my encoder interrupts from the timer and use my own flags ( I already am anyway on the encoder change function ), at this point I am just better off not using ioabstaction.

Another thought I had was to be able to pause or disable some delegates in real time, either way I am not going down the rabbit hole of threading or multi core, I just wanted to see if I can use ioabstaction and taskmanager decoupled from each other and get my callback right out of ioabstraction on interrupt without taskmanager mashalling it

davetcc, 2021-06-14:
> get my callback right out of ioabstraction on interrupt without taskmanager mashalling

To be honest this is what IoAbstraction goes to great lengths to avoid. If you need to run things directly in an ISR, I would avoid using libraries for that code and do that part yourself to ensure memory visibility, performance, and safety.

>Another thought I had was to be able to pause or disable some delegates in real time

That's actually where events come in handy, you can decide how long between executions, you can even safely trigger them remotely. But there are other ways, you can cancel tasks and re-instate them later, or you can do javascript setTimeout(..) style scheduleOnce where you call it again at the end of the timer event.

tablatronix, 2021-06-14:
I am using taskmanger in almost all my code bases , and am using 5-20 timers in each. But i am also using encoders in them all.. So I need some way to queue encoder priority over the other timers when its active otherwise responsiveness is not reliable, and since ioabstraction is tied to the same timer my options are limited, I was trying to figure out how to defer runloop or temporarily disable all other timers (obviously the smallest code change would be nice atm). I am also using ioabstraction because some use i2c and its nice having the abstraction already there and plan on using it more for other stuff as well.

The best way would be ioabstraction to have its own native timer.

second best modify all my code to add locking for all timer callbacks

Another way is to get encoder status manually from ioabstraction without timer loop, and then defer or skip timer loop. Not sure how bad this actually is, nothing is timing critical and not sure how well taskmanager handles delays...

davetcc, 2021-06-14:
In terms of within runLoop() the interrupt handler already takes the highest priority, checked first on every iteration. One thing I keep thinking through is moving the rotary encoder into an event, giving far more control over it and giving the option to do the counting inside the interrupt. But this can also be very dangerous, as not all platforms would handle this gracefully, especially when using Arduino's digital read. However, it *may* be OK in ESP32 extra mode as we use the IDF gpio functions, but potentially problematic on others.

I suspect much of the problem is to do with the synchronous nature of the Arduino API, specifically in many cases the way I2C works. There is no real way around it without fixing each library up in turn. For example, if we use u8g2 in tcMenu then we always provide our own data callback, which sends a bit of data over I2C and then yields to task manager. We will try to overcome this for some boards in the next version of IoAbstraction on ESP32 and mbed. But using platform I2C is very manual and per platform, so would only be supported for the ones we use a lot.

davetcc, 2021-06-14:
I mean we could probably provide a enable/disable flag on the tasks, so you could do something like:

`taskManager.enableTask(true|false);`

Then when you enabled it, it would just schedule itself again.

tablatronix, 2021-06-14:
That is also what I was thinking, I can just temporarily disable all high load tasks if i need too without having to modify all the callbacks themselves. Heck I could even disable all network tasks if wifi is not available without having to add wifi status checking to every single task.. very interesting

davetcc, 2021-06-14:
Add the following two new methods to taskManager:

```void enableTask(taskid_t taskId, boolean taskEnabled);
boolean isTaskEnabled(taskid_t taskId);```

We'll have a new status flag for enabled, I've checked and there are enough spare bits in the current execution mode status flag to deal with this.

* On enable, the task will be put back into the running list and the next scheduled time calculated. This means that the task would continue as normal. 
* On disable, the flag would be set to mark it as such, it would then not even need to be removed from the queue, calling execute would do nothing, and it would not repeat. 

This would not apply to events, as they are often triggered from interrupts, and the user has complete control over when events execute anyway. It would apply to functional and OO (executable) tasks.

davetcc, 2021-06-14:
See https://github.com/davetcc/TaskManagerIO/issues/30

davetcc, 2021-10-03:
The work in taskmanager is now being tested, working on ESP8266 already. Once tested on more platforms I'll put it onto master.

https://github.com/davetcc/TaskManagerIO/tree/enable-task-work

closed 2021-10-03 by davetcc