TcMenu/IoAbstraction #135: When navigate in TCMENU with buttons, analog read is impacted

kitsen13, 2021-10-18:
Hello,

I have a pressure meter on PIN 17 on ESP32.
It works if i'm not navigating / playing with button of the menu.

Navigate in TCmenu have a significant impact of the value of the ADC reading.

Please look at attached. (blue is raw measure, red is a filtered measure)

[image]

kitsen13, 2021-10-19:
Hello,
I 'm using in my project : tcmenu, taskmanager and IoAbstraction.
All the hardware devices (without sensor are described here : [](https://www.thecoderscorner.com/jforum/posts/list/140.page) )

One problem is the value read by the ESP32 (Pression in my code below) is affected when I press a button (all buttons are concerned) on the keypad k845037. The value decreases. 
An other problem concerns noise generated on the value. The sensor output is very stable (checked with an oscilloscope) but not the reading. Is there a common 'point' between the two problems?

Please take a look at the simplest sketch to reproduce the problem  :

`#include "tcmenu_menu.h"
#include <ArduinoEEPROMAbstraction.h>
#define DEBUG  // A déclarer si un des debugs si dessous
//#define DEBUG_BOUTON
#define DEBUG_ANA
//#define DEBUG_DEBIT

// Caractéristiques débitmètre YF-S402 G1 / 4 DN8 0.3-6L / Min DC4.5V - 4380 impulsions de sortie pour 1 litre

// Gestion de KEROZEN

// Debut AMA - gestion des ANALOGIQUES

    // https://github.com/davetcc/IoAbstraction/blob/master/examples/analogExample/analogExample.ino
    #include <IoAbstraction.h>
    #include <TaskManagerIO.h>
    #include <DeviceEvents.h>
    #include <Wire.h>
    
    #define ANALOG_IN_PIN_PRESSION 34 // Capteur de pression 0-100 PSI mais bridé à 66 PSI car 3,3 v max
    #define FLOW_PIN_DEBIT 17 // Capteur de debit par impulsion 
    //#define ANALOG_IN_PIN_BOUTTON 35 // Pin des bouttons de MENU
    
    //Debut AMA - variable gestion débit 
    float Pression;
    float PressionFilt;
    //Fin AMA - variable gestion débit 
    
    AnalogDevice* analog = internalAnalogIo();

// Fin AMA - gestion des ANALOGIQUES

// Debut AMA - gestion des Interruptions

// Fin AMA - gestion des Interruptions

// Debut AMA - gestion des BOUTTONS

      #include <DfRobotInputAbstraction.h>
      #include <AnalogDeviceAbstraction.h>
      
      const PROGMEM DfRobotAnalogRanges myRanges {
        0.0001F, 0.2666F, 0.5363F, 0.8030F,  -2.9000F
      }; // right, up, down, left, select
      
      DfRobotInputAbstraction dfRobotIo = DfRobotInputAbstraction(&myRanges, 35, internalAnalogIo());
      IoAbstractionRef dfRobotKeys = &dfRobotIo;
      
      
// Fin AMA - gestion des BOUTTONS

void setup() {

      // DEBUG MODE - start up the serial port in a way compatible with 32 bit boards.
                      
      #ifdef DEBUG
        while (!Serial);
        Serial.begin(115200);
      #endif 

      // Debut AMA - gestion des Interruptions / Débit
[... 116 lines cut ...]

// Fin AMA cycle

// Début Fonction mise à l'echelle
    float mae(float x, float in_min, float in_max, float out_min, float out_max)
    {
      float out;
      out = ((x - in_min)*(out_max - out_min) / (in_max - in_min) + out_min);
    
      if (x <= in_min)
      {
        out = out_min;
      }
      
      if (x >= in_max)
      {
        out = out_max;
      }
      
      return out;
    }
// Fin Fonction mise à l'echelle
// Fin AMA`

Regards,

davetcc, 2021-10-19:
Many thanks for confirming. Please can you close the issue.