TcMenu/IoAbstraction #141: Support for multiplexer

loczi1, 2021-12-28:
Dear Dave, the library is awesome!
I didn't find implementation for the 74HC405x analog Multiplexer / Demultiplexer, other type of multpiplexers are also welcome.
If you won't implement it, plase provide some kind of example code how to use it in your library.
Thanks, Akos

davetcc, 2022-01-01:
Sorry, we don't have support for those devices at the moment. Building in support for such a device would not be difficult for anyone with moderate C++ experience. It would basically come down to another implementation of BasicIoAbstraction - you can see all the instances in the reference docs: https://www.thecoderscorner.com/ref-docs/ioabstraction/html/class_basic_io_abstraction.html

The nearest one to what you want to do could be the DfRobotInputAbstraction: https://github.com/davetcc/IoAbstraction/blob/master/src/DfRobotInputAbstraction.h

The simplest one is probably the NegatingIoAbstraction

davetcc, 2022-10-27:
We'll take a look at this to see how easy it is and add at least one multiplexer.