TcMenu/IoAbstraction #143: type uint8_t used as boolean

fractalclockwork, 2022-01-22:
In "KeyboardManager.cpp", MatrixKeyboardManager::exec() uses the return value of uint8_t ioDeviceDigitalRead() as a boolean.
...
if(!ioDeviceDigitalRead(ioRef, layout->getRowPin(r))) {
...
This causes side-effects in the keyboard manger when tested with PCF8574IoAbstraction() on Ardunio Uno.

davetcc, 2022-01-25:
I'm sorry, but you'll need to be clearer than "causes side-effects". Please can you provide the exact issues that you see so we have a more detailed description of the problem to work with.

It may even be that there are issues using the keyboard class with the PCF8574 due to the way that inputs tend to work on that device, as per the docs, IoAbstraction always registers inputs as input pull-up on this device, as otherwise, they would be floating.

fractalclockwork, 2022-01-25:
Thanks for reviewing this.
I've reverted my change,
...
            if(!ioDeviceDigitalRead(ioRef, layout->getRowPin(r))) {  //careful ioDeviceDigitalRead is not a bool
            //if(ioDeviceDigitalRead(ioRef, layout->getRowPin(r)) != 1) {
...
but unfortunately can no longer reproduce the issue, which exhibited itself as a key held.  Totally agree that's the type of symptom a missing pull-up.  Now that the circuit is on a perf-board of course it works as designed. 

Anyway, I really like how this library is organized and thank you for your work.
Cheers!

davetcc, 2022-01-25:
Thanks, I guess the best thing is to close the issue for now, and open it again if you see the issue again.

fractalclockwork, 2022-01-25:
That's fine for now.  Let's close this issue.

fractalclockwork, 2022-01-25:
Unable to reproduce

closed 2022-01-25 by fractalclockwork