TcMenu/IoAbstraction #146: Method to remove a switch and all callbacks.

apoxa, 2022-03-02:
I started playing around with your library and was wondering if it's possible to remove a switch and later assign a new callback. Another option would be the possibility to overwrite the callback with another one, including an empty method.

I poked around the SimpleCollection library and there is `removeByKey` which maybe can be used for this

davetcc, 2022-03-02:
Yes that remove method is new. It was added in the last release. It does lead the way for implementing this. It should be quite trivial now.

davetcc, 2022-03-03:
**Removal:**

The only slight difficulty with removal would be to be able to remove any interrupts attached, for device pins that would be done with detatchInterrupt, for pcf23017 we could probably implement that function. However, for pcf8574 abstractions it maybe not be that possible to detach it as it is shared across pins.

One option would be to leave the interrupt de-registration with the user, as it would be safer.

**Change callback:**

This could be done by just providing a new method on `SwitchInput` to replace the callback.

davetcc, 2022-03-03:
I've been thinking about this a bit more, I can see why we'd want to change the callbacks, but I really can't see why we'd want to remove rotary encoders or buttons at runtime (other than maybe for testing purposes) but maybe I'm missing something. I mean these are pieces of hardware that are wired to pins on a device. Unless you're implementing some kind of plug-and-play system where people could somehow rewire it at runtime.

Changing the callback on an encoder should be quite simple and similar to what was discussed for changing a key callback. However, removing a rotary encoder at runtime will have side effects as it involves interrupt registration.

apoxa, 2022-03-03:
Ok, I might have gotten the terminology wrong.
Maybe I can clear it up with my actual use case. I have a mechanical keypad with an encoder (JC Pro Macro 2). The idea is, that you can setup multiple predefined modes, where each key or the encoder does something different. So for example in mode 1 it changes volume, in mode 2 it acts as a scrolling or zooming tool.
So what I want to do is not physically remove any switches or the encoder, but change the callback to a new one or clear the callback so it does nothing.

davetcc, 2022-03-04:
Many thanks for clearing that up. Ok that is really easily dealt with by adding a couple of new methods.

Essentially adding onPressed to switches and changeCallback on the encoder interface. Switches already handles null cases, just need to ensure encoders do the same.

davetcc, 2022-03-04:
I’ll also handle remove with a note that in interrupt mode the interrupt handlers need to be dealt with separately. Who knows someone may want that case.

closed 2022-04-03 by davetcc