TcMenu/IoAbstraction #150: Write 2 byte value to eeprom shows sporadic error

loczi1, 2022-03-20:
I use the [write16](https://github.com/davetcc/IoAbstraction/blob/4019986e20bf84730ad4c090b3af81d014751b51/src/EepromAbstractionWire.cpp#L51) function to write integer (65278) value to the eeprom 24c25.
I can write and read back 1 byte value sucessfully, but not 2 byte.

In my teat code I write the 65278 to an address and when I read back a 8bit value so I can check wich 1 byte part is bad.
`void testWrite2EEPROM(int start_address )
{
     for(start_address; start_address<350; start_address+=2)
    {
        
        I2CEeprom.write8(start_address, (int) 65278);
        Serial.println(I2CEeprom.hasErrorOccurred() ? "Write failure" : "Write success");
        Serial.printf("Address:  %d -> Read value: ",  (start_address));
        Serial.print(I2CEeprom.read8(start_address), DEC);
        Serial.print(" - ");
        Serial.println(I2CEeprom.read8(start_address+1), DEC);
    }
}`
[log.txt](https://github.com/davetcc/IoAbstraction/files/8311933/log.txt)

I call _testWrite2EEPROM(60);_ and nothing else.
You can check my log as it is showing from address 244 it is always getting back the 2 byte info.

It seems for me it is sporadic issue but I want to resolve it, so do you think this is code issue (and this is a bug) or hardware issue?
I also can check the I2C with logic analyzer if the writing need to be checked.

davetcc, 2022-03-21:
Thanks for the test sketch, I have an AVR and a SAMD device with AT24 ROMs connected that I often test with, I'll try that sketch on it later and feedback.

loczi1, 2022-03-21:
I tried on Esp8266, but if needed I can test on Esp32.

davetcc, 2022-03-21:
I'll try it on an ESP8266 as well, I have one to hand that I can quickly attach an I2C EEPROM.

davetcc, 2022-03-21:
I wonder if it is bus speed. What speed are you running the bus at? I used the default which would be 100khz. There is code to protect against writes while the device is busy, I will check the read code for read after write.

loczi1, 2022-03-21:
I also use the default, as I don't modify anywhere. Just define the instance same as you did: 
_I2cAt24Eeprom anEeprom(0x50, PAGESIZE_AT24C128);_
Protection would be good idea.

davetcc, 2022-03-22:
I've run some more tests this morning with a new sketch with far less timing delays in it, set Serial baud to 115200, moved the print around to remove any delay between reading and writing. I still cannot reproduce with an AT24LC128 device on an ESP8266 board. Further, I checked the read, it does correctly wait for the device to respond before starting, but that is only really relevant on array writes not single writes. I would test your device in isolation first, to rule out wiring problems or maybe even a faulty device.

```

#include <IoAbstractionWire.h>
#include <EepromAbstractionWire.h>
I2cAt24Eeprom anEeprom(0x50, PAGESIZE_AT24C128);

void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);
  Wire.begin();

    for(int start_address=0; start_address<2048; start_address+=2) {
        uint16_t toCheck = rand() * 65000;
        anEeprom.write16(start_address, toCheck);
        uint16_t readBack = anEeprom.read16(start_address);
        Serial.printf("Address:  %d -> Write: %04x Read: %04x", start_address, toCheck, readBack);
        Serial.println();
        if(anEeprom.hasErrorOccurred()) {
          Serial.println("Write failure");
        }

        if(readBack != toCheck) {
          Serial.println("Mismatch detected");
        }
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

[test_output.txt](https://github.com/davetcc/IoAbstraction/files/8322221/test_output.txt)

loczi1, 2022-03-22:
Thanks for this compex test Dave! I have executed only this sketch and no Mismatch showed, So the device and the setup is working. However calling your sketch as a function from my project still showes Mismatches, for further investigation I will plug on a logic analyzer tomorrow.
Thanks for your effort!

davetcc, 2022-03-23:
Agreed, I think at this point we may need a dump of the I2C bus from the logic analyzer to see what's going on. I would suspect either one of the devices is not behaving properly, or there's strange behaviour somewhere when multiple I2C operations are involved. If you don't want to share the dump publicly, you can send me a private message in the forum.

loczi1, 2022-03-29:
Long wireing caused capacity issue. The eeprom is a sensitive device and sometimes didn't recognised a write session and didn't acknowledged it.

Thanks for your help Dave.
Issue can be closed.

closed 2022-03-29 by loczi1