TcMenu/IoAbstraction #157: Support for PCF8575C 16-bit I2C I/O expander

vzahradnik, 2022-07-06:
Hi,

In my Arduino project I use PCF8575C expander and I'd like to integrate it with your library without any hacks. Currently I use the following library:
https://github.com/4-20ma/I2cDiscreteIoExpander

I could implement a wrapper around the library with your `IoAbstractionRef` abstraction, however, the cleanest approach seems to me to add direct support into this library, similarly to PCF8574.

Any guidance is highly appreciated. For starters, I'd like to know which repositories need to be modified. This one for sure, but I'd like to have full support, including TcMenu builder.

Thanks!

vzahradnik, 2022-07-08:
> It looks like this should be a very trivial task. The PCF8575 basically acts like two PCF8574. The protocol is exactly the same. I will have a go at adjusting the current PCF8574 class to support both. It should be nothing more that changing to class to send and receive either one or two bytes depending on device.

Is there anything I could help with? I need the support now so I'm prepared to code it by myself and contribute back. If you point me to what you'd like to change and how, I can start working on that.

davetcc, 2022-07-08:
It's trivial enough that I'll have a go at adding the support and if you can test it with the device that would be great.

vzahradnik, 2022-07-08:
Sure, I will happily do that :)

davetcc, 2022-07-08:
You can look at the two files I've committed (that are shown in the issue), I don't have the device so can't test it, but I did test it with a PCF8574 and it should work in theory.

To use it in TcMenu just set the expander type in designer to custom for now and declare the expander with the same name in your sketch file. See the online help button on the expander editor for more on how to do that.

In the next tcMenu release I'll add it properly so at that point you wouldn't need the custom code.

`inline IoAbstractionRef ioFrom8575(uint8_t addr, pinid_t interruptPin = 0xff)`

vzahradnik, 2022-07-08:
Thanks, I will test that tomorrow.

vzahradnik, 2022-07-09:
@davetcc your code works perfectly. It's ready to be merged.

closed 2022-07-11 by davetcc