TcMenu/IoAbstraction #161: esp32-s3 dont have DAC

Xekz, 2022-08-02:
esp32-s3/IoAbstraction/src/esp32/ESP32AnalogDevice.cpp:11:10: fatal error: driver/dac.h: No such file or directory

davetcc, 2022-08-02:
Apologies but we don't support that board at the moment, we support S2 but we have not got an S3 board to test with. We'll schedule the work to support it though.

davetcc, 2022-08-02:
Any pointers on a basic dev board would be helpful.

davetcc, 2022-08-03:
I’ve ordered an esp32s3 mini as they look interesting. I’m sure like the s2 there will be a lot of incompatibility early on.

davetcc, 2022-08-14:
Fix now on master, tested on an unexpected maker tiny S3. IoAbstraction and TaskManager tests all pass.

davetcc, 2022-08-14:
Tested right through to menu creation in tcMenu, all working with encoder too.

davetcc, 2022-08-15:
Closing as smoke tested on ESP32, ESP32S2 and ESP32S3

closed 2022-08-15 by davetcc