TcMenu/IoAbstraction #163: Support for I2C EEPROMs smaller than 24C32 (32 Kbit)

brycecherry75, 2022-08-09:
Other than the I2C Slave Address, I noted that your library writes two additional address bytes which is only supported on 24C(64/128/256/512/1025) devices but is not compatible with 24C(01/02/04/08/16) devices which use only one additional address byte; if two additional address bytes are used for writing to a device which uses only one additional address byte, the contents of the second address byte (LSB) will be written to the EEPROM location LSB corresponding to the first address byte (MSB) when not wanted.

davetcc, 2022-08-09:
I’ll order a few sizes and make the change once they arrive.

davetcc, 2022-08-10:
Ok I’ve read the data sheets and understand how these work now. Not particularly complex but a bit messy. The top of the memory address masks out the LSB  of the I2C address. I’ll order a few devices for testing.

davetcc, 2022-08-12:
Tested with 24C02, 24C04, 24C08, 24C16, 24C128, all sizes worked as expected and test all the major cases

davetcc, 2022-08-15:
Closing after further testing ready for release.

closed 2022-08-15 by davetcc