TcMenu/IoAbstraction #177: serLevelEnabled

umsmin, 2022-11-19:
I try to start with tcMenu but getting on every example in the Arduino IDE 1.8.13 the following error:

`SimpleTest.cpp:98:41: error: 'serLevelEnabled' was not declared in this scope
         if(serLevelEnabled(SER_IOA_DEBUG)) {
`

The Library versions are:

tcMenu                   2.5.0
IOAbstraction         2.5.0
LiquidCristalIO       1.4.3
TaskManagerIO      1.3.5
SimpleCollections  1.2.3

The 
TcMenuDesigner   2.5.0

 It's for the ESP32 and the TFT_eSPI Library from Bodmer. This is only shown when I use the examples. 
On my own Menu only "Rotary Encoder" and "2x8 LCD".

davetcc, 2022-11-20:
Thanks for the report, it is because IoLogging is turned off, I'll fix this before the release. For now if you want to continue using this version set build flag IO_LOGGING_DEBUG and it would work.
 
I think you're using the developer version of IoAbstraction (2.5.0), Hopefully, this wasn't published anywhere yet. Did you take the master branch or was a package manager serving this up? Quite important, as it shouldn't be released yet as it is effectively untested.

davetcc, 2022-11-20:
TcMenu latest is 2.5.0
IOAbstraction latest should be 2.4.3 to use the tested version, which if you're new to it I really recommend

umsmin, 2022-11-20:
Thanks for your replay.
I get all the stuff directly from Github. I will better downgrade it and have a try.

davetcc, 2022-11-20:
FIX: ensure before next release that this compiles without the debug flag turned on

umsmin, 2022-11-20:
Now I tried it and it works without the error. But I can't make my own with the Bodmer libraries. I create a simple menu for touch but when I start to "generate code" it will not show me the eSPI option. Where is my mistake?

davetcc, 2022-11-20:
Make sure the platform in code generator (top section of the generator dialog)is set to one of :

ARDUINO_32
ARDUINO_ESP8266
ARDUINO_ESP32
STM32DUINO

These are the only devices that the underlying library supports

umsmin, 2022-11-20:
Yes it's on 

ARDUINO_ESP32

In your examples you don't have a touch menu with eSPI library from Bodmer?

davetcc, 2022-11-20:
We have plugin for Bodmers' TFT_eSPI display, but we've tended to use other input libraries. There are a few touch plugins, one for directly managing resistive touch input, and then another for common touch libraries.

umsmin, 2022-11-20:
How can I use it. The Bodmer are for me at this moment the best working ones. How can I add plugins for the different libraries?

davetcc, 2022-11-20:
I mean you could take a look at how we do the other touch plugins and try to write something similar for that library. Effectively you just need to plugin the X, Y and Z(force) into the IoAbstraction touch facilities. They are documented to some extent here: https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/touch-screen-support/

Effectively, you'd want to take a look how the two existing touch plugins work, and write your own touch interrogator as per the docs above. You could either do this as a plugin, or alternatively set the input code generator to "No Input" and just write your own code for it in setup.

Writing a plugin is quite straightforward. It's in XML format, here: https://github.com/davetcc/tcMenu/tree/master/xmlPlugins

umsmin, 2022-11-20:
Okay thank. But your plugins for the displays are not shown in the tcMenu Designer. Only the LCD-plugin. where is my mistake I made?

davetcc, 2022-11-20:
The only reason I can think of would be the platform is not right. Have you clicked on the image to select a different plugin?

umsmin, 2022-11-20:
Yes sure. 
As input is the rotary encoder.
As display is the LCD shield on I2C.
And for remote is no remote capability.

Must I put the plugins in a sub folder from designer?

davetcc, 2022-11-20:
There should be nothing to do other than select the plugins you want to use, those sound like the defaults, have you read through the getting started guide? Here's documentation about the code generator:

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/code-generator-and-plugins-guide/

I misunderstood before and thought you were asking how to write your own plugin, if all you want is to use the TFT_eSPI plugin, just click the change button on the left of the display plugin and change it to the one you want. Code generator will do the rest. Apologies for complicating things.

umsmin, 2022-11-20:
Where must be the plugins located? I don't see them anywhere. Not in the tcMenu Designer program folder or any sub folders. Also not in the Arduino -> libraries -> tcMenu folders.

The page behind the link I read and know how it works, but the plugins will never been shown.

Yes I only want to use it, don't make own plugins at this moment. I have an application and my menu system is unusable so I looked for a good one on the touch display.

davetcc, 2022-11-20:
This guide explains how to get started 

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/code-generator-and-plugins-guide/

umsmin, 2022-11-20:
Yes I know and read it not only once. But there is not shown where are the plugins located.

davetcc, 2022-11-20:
Ok by default the plugins are installed into the .tcmenu/plugins folder from your home directory. They would normally be put there at first start up by the UI.

They must be there as you wouldn’t even see any plugins at all otherwise. The package we create has all the plugins packaged with it.

umsmin, 2022-11-20:
They all there. I start the tcMenu Designer as normally user or administrator but nothing helps. He don't use them

davetcc, 2022-11-20:
If you can send your logs (in that .tcMenu directory under logs)  I’ll take a look at them tomorrow. There must be something strange going on there.

umsmin, 2022-11-20:
The only thing that's not working is the touch from the eSPI library display. The menu is now shown but wich input plugin can I use?

davetcc, 2022-11-20:
No problem I wanted to make sure there wasn’t a problem somewhere. For the input plugin you can either use the Arduino library touch plugin that supports many common libraries and touch interfaces. Or you can use the resistive touch plugin if you have the X and Y connectors directly connected.

davetcc, 2022-11-20:
If you’re using master of IoAbstraction you could pull the latest, I think I fixed that already.

umsmin, 2022-11-20:
I use this display 

[https://www.amazon.de/gp/product/B08SBTBRZV/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1](url)

and it works fine with eSPI library from Bodmer. 
The resistive touch plugin don't work because it has his own controller. 

The display is ILI9341
Touch is XPT2046

Adafruit libraries are not working.

It use the SPI interface for display and touch together. Only the CS_DISP and CS_TOUCH are on separate pins.

davetcc, 2022-11-20:
Yes the "FT6206/XPT2046 touch screen plugin that provides integration with library" plugin should work, as long as you install the XPT2046 library by Paul Stoffregen. It just needs the CS pin to be defined in order to work.

closed 2022-12-17 by davetcc