TcMenu/IoAbstraction #178: Interrupts not working?

tablatronix, 2022-12-04:
Strange issue I just noticed not sure if by design..

I am using 

`setupRotaryEncoderWithInterrupt` but I noticed my callbacks only work if I run taskmanager loop, which seems odd. 

I thought I posted this before but couldn't find it.

tablatronix, 2022-12-04:
#12 

Not sure what I am doing wrong or how to check this

tablatronix, 2022-12-04:
Ok it looks like I still need to separate my switch timer from my taskmanager tasks so I can faster poll my encoder, or defer my tasks. I think I remember bringing this up a while ago, but forget where I landed on it...

davetcc, 2022-12-07:
Yes, this is as per design, you can't really do anything in an ISR, other than flag that something has happened and deal with anything safety critical.

It's difficult with Arduino sometimes because much of the API is synchronous, although there are usually asynchronous options on each platform. The biggest offender for this is usually I2C because it is very slow. Normally, the slowness is caused by waiting for hardware.

davetcc, 2022-12-07:
In fact, on that front, as of the latest release interrupts for encoders are now optional. The only real benefit is that you can avoid polling, so it makes it easier to add a power management library and put the board into a low-power state between tasks (as there are no polling tasks).

tablatronix, 2022-12-07:
I will look into how to use multiple taskmanager queues instead or add locks to every task somehow.

closed 2022-12-07 by tablatronix