TcMenu/IoAbstraction #179: insufficient documentation causing compiler-errors

StefanL38, 2022-12-26:
I installed the IO-abstraction library with the Arduino 1.8.19-ide
with all the mentioned other dependend libraries 
[image]

I took the example-code from here as raw text
https://raw.githubusercontent.com/davetcc/IoAbstraction/master/examples/rotaryEncoderLed23017/rotaryEncoderLed23017.ino
Crtl-A- Ctrl-C
pasting Ctrl-V into an empty sketch and compiled
compiler-errors

´´´
Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Arduino Uno"

E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\hardware -hardware C:\Users\dipl-\AppData\Local\Arduino15\packages -tools E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\tools-builder -tools E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\hardware\tools\avr -tools C:\Users\dipl-\AppData\Local\Arduino15\packages -built-in-libraries E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\libraries -libraries F:\myData\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\dipl-\AppData\Local\Temp\arduino_build_814318 -warnings=all -build-cache C:\Users\dipl-\AppData\Local\Temp\arduino_cache_400908 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose F:\myData\Arduino\rotaryEncoderLed23017.ino\rotaryEncoderLed23017.ino.ino

E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\hardware -hardware C:\Users\dipl-\AppData\Local\Arduino15\packages -tools E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\tools-builder -tools E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\hardware\tools\avr -tools C:\Users\dipl-\AppData\Local\Arduino15\packages -built-in-libraries E:\Arduino-funktioniert\arduino-1.8.19-windows-2022-01-30\arduino-1.8.19\libraries -libraries F:\myData\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\dipl-\AppData\Local\Temp\arduino_build_814318 -warnings=all -build-cache C:\Users\dipl-\AppData\Local\Temp\arduino_cache_400908 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose F:\myData\Arduino\rotaryEncoderLed23017.ino\rotaryEncoderLed23017.ino.ino

Using board 'uno' from platform in folder: C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...

"C:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_814318\\sketch\\rotaryEncoderLed23017.ino.ino.cpp" -o nul

Alternatives for IoAbstraction.h: [IoAbstraction@2.4.3]

ResolveLibrary(IoAbstraction.h)

  -> candidates: [IoAbstraction@2.4.3]

"C:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IF:\\myData\\Arduino\\libraries\\IoAbstraction\\src" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_814318\\sketch\\rotaryEncoderLed23017.ino.ino.cpp" -o nul

Alternatives for TaskManagerIO.h: [TaskManagerIO@1.3.5]

ResolveLibrary(TaskManagerIO.h)

  -> candidates: [TaskManagerIO@1.3.5]

"C:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IF:\\myData\\Arduino\\libraries\\IoAbstraction\\src" "-IF:\\myData\\Arduino\\libraries\\TaskManagerIO\\src" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_814318\\sketch\\rotaryEncoderLed23017.ino.ino.cpp" -o nul

Alternatives for SimpleCollections.h: [SimpleCollections@1.2.3]

ResolveLibrary(SimpleCollections.h)

  -> candidates: [SimpleCollections@1.2.3]

"C:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IF:\\myData\\Arduino\\libraries\\IoAbstraction\\src" "-IF:\\myData\\Arduino\\libraries\\TaskManagerIO\\src" "-IF:\\myData\\Arduino\\libraries\\SimpleCollections\\src" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_814318\\sketch\\rotaryEncoderLed23017.ino.ino.cpp" -o nul

Alternatives for Wire.h: [Wire@1.0]

ResolveLibrary(Wire.h)

  -> candidates: [Wire@1.0]

"C:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\dipl-\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IF:\\myData\\Arduino\\libraries\\IoAbstraction\\src" "-IF:\\myData\ [... cut ...]
Bibliothek TaskManagerIO in Version 1.3.5 im Ordner: F:\myData\Arduino\libraries\TaskManagerIO  wird verwendet

Bibliothek SimpleCollections in Version 1.2.3 im Ordner: F:\myData\Arduino\libraries\SimpleCollections  wird verwendet

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\dipl-\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire  wird verwendet

exit status 1

expected constructor, destructor, or type conversion before '(' token
´´´
You should add information with which type of microcontroller this works
or remove the bug

best regards Stefan

davetcc, 2022-12-27:
This is actually because there was a mismatch between what was on the main branch in the repo. I've released the library now, but it is always better to ensure you use the example for the version of the library you're using.

davetcc, 2022-12-27:
Version 3.0 was released this morning so GitHub and the library version should now be in sync. I still recommend you use the examples that are packaged with the library however.

davetcc, 2022-12-27:
And on the documentation front, we're open-sourcing the documentation on TheCodersCorner soon, and any suggestions on improvements would be most welcome.

davetcc, 2022-12-28:
Thanks for testing this and reporting these issues in such detail. I've released 3.0 that fixes the original bug, and then 3.0.1 that fixes up the residual errors that we found later. Can this be closed now or do you think there are more things left to do?

davetcc, 2023-01-03:
Thanks again for helping out, and for proving so much detail on the forum. The reverse direction prevention logic has been added to the debouncing in V3.0.

I think you just happened to pick an unfortunate time to try the library when a major upgrade was taking place, the documentation was also in flight. Adding to this the example itself was not particularly good. All these issues have since been corrected, and  github is back in sync with the what's released.

I'll close this, but  if you think there's still outstanding problems please re-open.

closed 2023-01-03 by davetcc